MikePopoloski/slang #137: Question about using slang to verify code.

oddball, 2017-09-06:
This is more of a discussion type ticket.

What sparked my interest in this software, was that I perhaps could use it on CircleCI to test that the files generated by my ipxact2systemverilog tool generates correct SystemVerilog. Specifically these files:

https://github.com/oddball/ipxact2systemverilog/blob/master/example/output/example_sv_pkg.svh
https://github.com/oddball/ipxact2systemverilog/blob/master/example/output/example_sv_pkg.sv

Is that a reasonable assumption?

I can't see an executable generated by slang that I could use. Is this meant a library only or will there be an executable around.

MikePopoloski, 2017-09-06:
Yes, this is something you should be able to do. Right now there is only a library, but eventually there will be various executables. Similar to libclang or Roslyn, the compiler is being built as a library / service first and foremost, which makes it easy to adapt into whatever tool you might need.

Obviously things are still under very heavy development, but the lexer / preprocessor / parser stack is pretty solid. If you look at the depmap tool that I started working on, you can see an example of parsing every SystemVerilog file in a given set of directories and printing out any errors encountered.

martinda, 2017-09-06:
If I may, @MikePopoloski what are your expectations or desires in terms of contributions to this project. I too noticed there was no executable, and I pondered, if my time permits, whether or not I should attempt to tackle Issue #125, or perhaps an easier issue (and learn C++ a the same time).

MikePopoloski, 2017-09-06:
I'm happy to accept contributions from anyone, with the understanding that I want to maintain a consistent style and a high level of quality (which I can help you achieve via code review).

#125 would be a good issue to tackle I think because it's fairly stand alone and the underlying library code is basically finished. There is a little bit of work to figure out how exactly the tokens should be converted to strings around macros (once the macro is expanded, do we keep comments? throw them away? what about white spacing before the macro? etc). There are flags that control some of these things right now but it could probably be expanded / polished.

MikePopoloski, 2017-09-29:
Going to close this out, I think the discussion has run its course.

closed 2017-09-29 by MikePopoloski