MikePopoloski/slang #168: Build fails on Mac OSX: unavailable posix_fadvise()

albertschulz, 2018-06-04:
I get the following error when compiling on Mac OSX. It seems that the code relys on the posix_fadvise function, which is not available in Mac OSX. Can we just leave out this function for Mac OSX? Do you know which other function could be used to achive the same functionality?
```
➜ slang git:(master) make -C build/projects/gmake-linux -j 8
==== Building slang (debug64) ====
format.cc
ostream.cc
posix.cc
printf.cc
ConstantValue.cpp
EvalContext.cpp
Expressions.cpp
Expressions_eval.cpp
Expressions_prop.cpp
Statements.cpp
Compilation.cpp
Diagnostics.cpp
Lexer.cpp
LexerFacts.cpp
Preprocessor.cpp
Token.cpp
SVInt.cpp
Time.cpp
VectorBuilder.cpp
AllSyntax.cpp
Parser.cpp
ParserBase.cpp
Parser_expressions.cpp
Parser_statements.cpp
SyntaxFacts.cpp
SyntaxNode.cpp
HierarchySymbols.cpp
Lazy.cpp
MemberSymbols.cpp
Scope.cpp
SemanticFacts.cpp
StatementBodiedScope.cpp
Symbol.cpp
TypePrinter.cpp
TypeSymbols.cpp
SourceManager.cpp
BumpAllocator.cpp
Hash.cpp
Path.cpp
Util.cpp
../../../source/util/Path.cpp: In member function 'bool slang::Path::readFile(std::vector<char>&) const':
../../../source/util/Path.cpp:112:29: error: 'POSIX_FADV_SEQUENTIAL' was not declared in this scope
     posix_fadvise(fd, 0, 0, POSIX_FADV_SEQUENTIAL);
                             ^~~~~~~~~~~~~~~~~~~~~
../../../source/util/Path.cpp:112:5: error: 'posix_fadvise' was not declared in this scope
     posix_fadvise(fd, 0, 0, POSIX_FADV_SEQUENTIAL);
     ^~~~~~~~~~~~~
../../../source/util/Path.cpp:112:5: note: suggested alternative: 'posix2time'
     posix_fadvise(fd, 0, 0, POSIX_FADV_SEQUENTIAL);
     ^~~~~~~~~~~~~
     posix2time
make[1]: *** [../../linux64_gcc/obj/x64/Debug/slang/source/util/Path.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [slang] Error 2
```

MikePopoloski, 2018-06-06:
posix_fadvise is just a hint to potentially speed things up; if it's not available on OSX it can just be removed.

MikePopoloski, 2018-06-07:
ifdefed it out

closed 2018-06-07 by MikePopoloski