MikePopoloski/slang #170: Sorry maybe not an issue but curious, how far along is this project, and how can i contribute?

jpeezzy, 2018-06-17:
Is it possible to run verilog code as of yet?

MikePopoloski, 2018-06-20:
No, no code execution yet. I'm making slow but steady progress on type checking and semantic analysis. I think fairly soon I will start on adding an LLVM backend for simulation.

If you're serious about contributing I could come up with a list of things to work on that help me out.

jpeezzy, 2018-06-24:
I'm a computer engineer, but i've never really dealt with compilers before. Do you have any textbooks that can help me get started? I like the idea of having a clang compiler for verilog simulations.

martinda, 2018-06-24:
@jpeezzy if I may, I'd like to suggest the two books from Terence Parr, [Language Implementation Patterns](https://pragprog.com/book/tpdsl/language-implementation-patterns) and [The Definitive ANTLR 4 Reference](https://pragprog.com/book/tpantlr2/the-definitive-antlr-4-reference). I have read and abandoned many book on compilers, but these two I have read many times.

MikePopoloski, 2018-06-26:
I don't have any personal recommendations for compiler books. Architecting a compiler is equal parts past experience and observing what others have done before. On the latter side of things, try looking at talks and blog posts from the Rust, Clang, or C# compiler devs.

martinda, 2018-06-26:
You could also look at the D programming language compiler source code.

MikePopoloski, 2018-07-19:
Rather than a skype call (which I don't have and maybe more importantly doesn't allow things like saving / searching history) maybe I should open a slack or discord channel and we can discuss it there?

MikePopoloski, 2018-07-19:
Opened a Gitter page for this repo here: https://gitter.im/MikePopoloski/slang

Please use that for further discussion.

closed 2018-07-19 by MikePopoloski