MikePopoloski/slang #259: Generating system verilog from scratch

Kuree, 2019-06-18:
Is there any plan to allow tools to build arbitrary AST trees from scratch and generate system verilog code? My impression is that the internal AST is geared towards parsing from text.

I'm happy to help if you think it's a good idea.

MikePopoloski, 2019-06-19:
This is intended to be possible via the syntax tree structures, though there's no API right now to make it convenient to do.

Doing it from the actual AST (symbol trees) is potentially not the right place; symbols are abstract enough from the actual syntax that it's hard to know how to translate between them. If you have ideas for this I'd be interested, but I think it more likely that tools know the code they want to generate, in which case they can just construct syntax trees appropriately.

Kuree, 2019-06-19:
Sorry I was referring to abstract syntax trees. Is there any easy way to construct syntax trees programmatically?

MikePopoloski, 2019-06-19:
Yes, you can instantiate a SyntaxFactory and then create any nodes that you need. Once you have a tree you can print it back to source via SyntaxPrinter. It's just not very ergonomic to create the tree right now, because you need to create each token, add trivia for whitespace, etc. Some helper methods to reduce the boilerplate would go a long way.

closed 2019-07-23 by MikePopoloski