MikePopoloski/slang #261: Parsing failure

renau, 2019-08-13:
It faults when I try to parse the VQM (verilog subset) from titan_release_1.3.0/benchmarks/titan23/directrf/netlists/directrf_stratixiv.vqm

The source is at http://www.eecg.utoronto.ca/~kmurray/titan.html

I tried ../build/bin/rewriter .directrf_stratixiv.vqm
and seg-faults

MikePopoloski, 2019-08-15:
tgorochowik fixed the immediate failure, which was an accidental truncation of the file size when over 4GB. However you'll run into other issues in different parts of the library which assume source files never go over 4GB (for example, all SourceLocations currently use a uint32_t for representing char offsets).

I believe those issues can be fixed without too much work, but a bigger problem is that it's going to require a machine with a *ton* of memory because slang often makes a tradeoff of using more memory in order to run faster. If you have a machine with, say, 32GB of RAM you probably won't have any trouble running the rewriter on a 4GB+ file, but 16GB of RAM might be cutting it too close. I bet there are a bunch of places that could be optimized for memory usage without harming perf too much but doing so has been pretty far down my priority list.

MikePopoloski, 2019-08-25:
This should be fully fixed now. SyntaxPrinter is still fairly slow so the rewriter test isn't particularly fast but that's not a primary use case anyway.

PR #262 (2019-08-15): Source manager memory allocation fix
Casting the total size to u32_t effectively limits the max files size. This can cause a SEGFAULT if the input file is larger than 2**32 B.
    
Fixes #261

Additionally this PR adds some sanity checks to the rewriter tool.
commit 96b0a003 (2019-08-15): Fix memory allocation in SourceManager

Casting the total size to u32_t effectively limits the max files size.
This can cause a SEGFAULT if the input file is larger than 2**32 B.

Fixes #261
closed 2019-08-25 by MikePopoloski