MikePopoloski/slang #263: sv-lang.com website in "Try It Out" doesn't work?

mithro, 2019-09-02:
The [Try It Out](https://github.com/MikePopoloski/slang/tree/master#try-it-out) section of the README.md links to `sv-lang.com` which appears to be down?

https://github.com/MikePopoloski/slang/blame/master/README.md#L21-L23
```markdown
### Try It Out
Experiment with parsing, type checking, and error detection live [on the web](http://sv-lang.com/) (this tool is inspired by Matt Godbolt's excellent [Compiler Explorer](https://godbolt.org/)). It's still pretty rough but useful for exploring the interactive behavior of the toolchain.
```

MikePopoloski, 2019-09-03:
It's back up now. I run it off an AWS instance but there's something misconfigured about it and it doesn't always stay up forever. I haven't had time to really look into it.

mithro, 2019-09-03:
Working for me now! It's pretty cool website.

closed 2019-09-03 by mithro