MikePopoloski/slang #264: New open source SystemVerilog compliance suite being created

mithro, 2019-09-03:
It might be of interest to your that the SymbiFlow project has started working on an open source System Verilog compliance suite which can be found at https://github.com/SymbiFlow/sv-tests

It's still very early stages, but its goal is to cover the whole of the System Verilog LRM. 

The project is currently trying to produce a "status report" for popular open source tools and your slang parser is included. The resulting output report can be found at https://symbiflow.github.io/sv-tests/.

MikePopoloski, 2019-09-03:
Nice, that's super cool. I spot checked a couple of the failures for slang and the ones I saw all seemed to be because the test file used a macro but never defined it anywhere? Not sure how useful you expect those results to be just yet but I think the idea is good.

mithro, 2019-09-03:
Please do log issues for things you find that don't make sense. It is still very much a WIP.

mithro, 2019-09-03:
FYI @MikePopoloski - The step after getting the test suite into a good state is figuring out which SystemVerilog parser to start contributing too to and integrate into other tools.

MikePopoloski, 2019-09-03:
Well I'm certainly happy to accept contributions. The library is designed to be easily integrated as a front end for any tool -- if you run into pain points with that please let me know.

closed 2019-09-15 by MikePopoloski