MikePopoloski/slang #265: sv-lang.com refused to connect.

kevbroch, 2019-09-14:
Was working a day or so ago:
```
sv-lang.com refused to connect.
Try:

Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED
```
Also have a suggestion could you put either static html version of it or screen capture of the matrix (maybe just partial I know it's big) on your github repo so if site is down people can still get the gist of it?

ps:  I love this!  Exactly what the OSS ecad tools need for SV

MikePopoloski, 2019-09-15:
Yeah, that's probably a good idea. At some point I'll look into why the server won't stay up.

MikePopoloski, 2019-10-05:
It's back up now. I think it was being killed by the OOM killer, because I'm running this on the cheapest possible AWS instance.

closed 2019-10-05 by MikePopoloski