MikePopoloski/slang #291: SystemVerilog source formatting tool

atkenny15, 2020-01-27:
Hi @MikePopoloski,

I am very interested in a `clang-format`-like tool for SystemVerilog and looking through the `rewriter` code, it looks like this project provides great support for building such a tool.  Before I start writing this, I wanted to talk to you and see if you had any plans for development in this area.

- Is this a tool that I should work on within the `slang` project, or should it be separate and just use `slang` as the library?

mithro, 2020-01-27:
@atkenny15 You should check out [Google's veribile](https://github.com/google/verible).

> ## Formatter
> The formatter is a transformative tool that manages whitespace in accordance with a particular style. The main goal is to relieve humans of having to manually manage whitespace, wrapping, and indentation, and to provide a tool that can be integrated into any editor to enable editor-independent consistency.

atkenny15, 2020-01-27:
@mithro Thanks for the pointer, that looks like it will solve my problem

mithro, 2020-01-27:
slang is pretty cool too -- Takes a look at how it goes in our sv-test suite.
 * https://github.com/SymbiFlow/sv-tests
 * https://symbiflow.github.io/sv-tests/

MikePopoloski, 2020-01-27:
Yes, slang's API is designed to make it very easy to write a clang-format kind of tool, I simply haven't had time to work on that. I'd be happy to accept such a tool into the repo if you wanted. If you're happy to just use verible that's fine, but if you still want pointers on slang's API I'd be happy to help.

atkenny15, 2020-01-28:
@MikePopoloski Thanks for the reply.  Verible's style is not what I'm looking for so it might be easier to add formatting support to slang.  I'm still evaluating the effort to add my coding style to Verible's formatter vs. adding a formatter to slang.  I appreciate your help and will let you know when I've decided on a direction.  Thanks again!

MikePopoloski, 2020-02-13:
I'm going to close out this issue; feel free to open another or message me on gitter if you decide to come back to this.

closed 2020-02-13 by MikePopoloski