MikePopoloski/slang #293: How to iterate within ProceduralBlockSymbol?

Kuree, 2020-02-27:
In `BlockSymbols.cpp`: 
```
void ProceduralBlockSymbol::serializeTo(ASTSerializer& serializer) const {
    serializer.write("procedureKind", toString(procedureKind));
}
```
Is there any way to iterative within the procedural block so get its content? Thanks.

Kuree, 2020-02-27:
Nevermind. I think I figured out why: proper expression statement symbol is not implemented. I will experiment more and hopefully send out a PR soon.

closed 2020-02-27 by Kuree

MikePopoloski, 2020-02-27:
Actually you can get the body of the procedural block by calling `getBody()` , which will give you a `Statement` tree.

The statement tree should be full functional. It's true that I haven't yet implemented JSON serialization for the Statement classes though, and would welcome a PR that adds those. If you want to look at how to do that you can follow the example for `Expression` trees in `ASTSerializer`

Kuree, 2020-02-29:
Thanks. It turns out adding symbols for each statement is way more work than simply implementing the serialization for statement classes.

I've sent a PR that adds `serializeTo` method to all statement classes. See https://github.com/MikePopoloski/slang/pull/294.