MikePopoloski/slang #300: Unexpected appending of IntegerLiterals

atkenny15, 2020-03-31:
Using the rewriter tool, it looks like IntegerLiteral values are being incorrectly appended to previous ones:

```
$ cat ../test3.sv 
reg [3:0] test;
reg [1:0] val;

always_comb begin
    casez (test)
        4'b???1: val = 0;
        4'b??10: val = 1;
        4'b?100: val = 2;
        4'b1000: val = 3;
    endcase
end
```

```
$ ./bin/rewriter ../test3.sv
reg [3:0] test;
reg [1:0] val;
always_comb begin
    casez (test)
        4'b???1: val = 0;
        4'b???1??10: val = 1;
        4'b???1??10?100: val = 2;
        4'b1000: val = 3;
    endcase
end
```

atkenny15, 2020-03-31:
This seems to fix it (although I'm not convinced that this is the correct way):

```diff
diff --git a/source/parsing/NumberParser.cpp b/source/parsing/NumberParser.cpp
index 26394564..18e83e7f 100644
--- a/source/parsing/NumberParser.cpp
+++ b/source/parsing/NumberParser.cpp
@@ -7,6 +7,7 @@
 #include "slang/parsing/NumberParser.h"
 
 #include "../text/CharInfo.h"
+#include <fmt/format.h>
 
 #include "slang/diagnostics/LexerDiags.h"
 #include "slang/util/String.h"
@@ -167,9 +168,10 @@ int NumberParser::append(Token token, bool isFirst) {
 
 Token NumberParser::finishValue(Token firstToken, bool singleToken) {
     auto createResult = [&](auto&& val) {
-        return Token(alloc, TokenKind::IntegerLiteral, firstToken.trivia(),
-                     singleToken ? firstToken.rawText() : toStringView(text.copy(alloc)),
-                     firstLocation, std::move(val));
+        const auto temp = singleToken ? firstToken.rawText() : toStringView(text.copy(alloc));
+        text.clear();
+        return Token(alloc, TokenKind::IntegerLiteral, firstToken.trivia(), temp, firstLocation,
+                     std::move(val));
     };
 
     if (!valid)
```

MikePopoloski, 2020-04-01:
Wow yeah, good catch. I made a similar fix and added a test that reproduces the issue. Thanks for the report!

closed 2020-04-01 by MikePopoloski