MikePopoloski/slang #304: Consider creating a UHDM compatible binding for slang?

mithro, 2020-07-24:
As part of the [Surelog project](https://github.com/alainmarcel/Surelog) (a /potentially/ competing ANTLR4 based SystemVerilog parser) and [Verible project](https://github.com/google/verible)  (a /potentially/ Bison based SystemVerilog parser focused on code linting and formatting) we have been trying to create a system which allows different parsers be interfaced to multiple different tools (like Yosys, Verilator, Icarus, etc). 

The attempt at doing this project is [called Universal Hardware Data Model (UHDM)](https://github.com/alainmarcel/UHDM) and structured kind of like this;
[image]

It would be awesome to see more parsers trying to support this API to make sure the API isn't over fitted to Surelog. 

I understand this is potentially quite a **huge** task so don't expect you to /actually/ do this but thought it could hurt to propose the idea.

FYI the related UHDM issue is at https://github.com/alainmarcel/UHDM/issues/319

MikePopoloski, 2020-07-24:
I'll definitely take a look, thanks for raising the issue.

Philosophically, slang is already designed to be the fastest (which it seems to be) and (eventually) the most complete parser (along with other front-end pieces) that all other tools can simply link to and do their backend work. Having a standard for the object model does seem (genuinely!) useful, but I'm also hoping people will ditch the auto-generated parsers and go with this one since it's faster and provides much nicer error messages, but I also realize that's not likely to make me friends in the short term. I'll look more into UHDM and see how I can incorporate it into that goal.

MikePopoloski, 2025-08-27:
I've checked in on this a few times over the years. It seems like UHDM never really went anywhere? It seems like downstream users like Yosys and CIRCT are going to use slang as a compiled frontend instead of having an intermediate layer like UHDM in the middle. If there are still real world users of UHDM, perhaps someone could contribute a UHDM serializer to slang? It's not something I have time to implement myself, but would be happy to support it if it would be useful to someone.

I haven't been able to find a roadmap or long term planning discussion around the future of Surelog and UHDM. It seems like maybe there are sporadic maintenance commits but it seems to me that a lot of wasted effort could be saved if Surelog committers just contributed whatever extra bits of functionality they need to slang (like UHDM serialization, for example) rather than trying to maintain a parallel implementation for a language as complicated as SystemVerilog.

@hzeller since I think you're still active in this space.

MikePopoloski, 2025-09-06:
Alright well, I'm going to close this since I'm not going to do this myself. If someone is still getting value out of UHDM it still makes way more sense to me for them to support a serializer within slang rather than trying to maintain a whole parallel frontend implementation but not my place to say obviously.

closed 2025-09-06 by MikePopoloski

heyfey, 2026-11-12:
@mithro @MikePopoloski @hzeller the separate project: https://github.com/heyfey/slang-uhdm