MikePopoloski/slang #305: Escape sequences in output on Windows

a2k-hanlon, 2020-08-05:
Hi there!

I'm trying slang for the first time. Thanks for all the work on it! I'm using the v0.3 Windows binary. When I run slang from PowerShell or the command prompt, there are escape sequences that get printed in slang's output.

For example, the command
```
$ slang -Weverything counter.sv
```
gives this:
```
[36mcounter.sv[0m:[96m14:12[0m: [91merror: [0m[1m'couter' doesn't match previous name 'counter'[0m
endmodule: couter
[92m           ^[0m
```

Is this something that I should address myself or might this be something to fix in slang? Is it an issue with detecting color-capable terminals?

I think this page provides a potential fix (for both my end and slang's end): https://stackoverflow.com/questions/51680709/colored-text-output-in-powershell-console-using-ansi-vt100-codes

Thanks!

MikePopoloski, 2020-08-05:
Thanks for the detailed report. It does look like I could fix this in slang by calling the right console method, so I'll play around with that. I do develop on Windows but I use ConEmu which natively supports the ANSI color codes, which is why I didn't notice this.

In the mean time, it looks like you could fix this on your end by either enabling ANSI color support via user settings, or you can tell slang to stop outputting them with the flag `--color-diagnostics=false`

MikePopoloski, 2020-08-05:
Should be fixed at HEAD

closed 2020-08-05 by MikePopoloski

a2k-hanlon, 2020-08-06:
Ok, good to hear! Thanks very much for the suggestions and the quick fix.

a2k-hanlon, 2020-08-10:
Hey @MikePopoloski, just popping back here to say thanks again for the help. You were right, I successfully enabled terminal colours through a registry setting as in that SO page, but your suggestion of the ```--color-diagnostics=false``` flag was what I needed to get slang working with the [linter-veriloghdl](https://github.com/a2k-hanlon/linter-veriloghdl) package for Atom! Without your fix within slang, that is :)

With your fix, it just works. Brilliant!

MikePopoloski, 2020-08-10:
Great, glad to help.