MikePopoloski/slang #330: Question about astvisitor

Winging, 2020-10-28:
Astvistor provides a base class for users  to visit ast tree. But  a class derives from Astvisitor may visit expressions or statements twice or more because  `StatementBlockSymbol` or `SubrountineSymbol`   derives from both symbol and scope  and they all have getBody() method.   In AstVisitor's visitDefault() method, it visits scope's members and body which result in multiple visit of ast.
I am wondering how to fix it, or whether astvisitor is not intended to be an interface to users?

MikePopoloski, 2020-10-28:
Hmm, that's a good point. It's an oversight, ASTVisitor is intended to only visit the AST once. The statement trees mirror block symbol trees when the blocks form part of the hierarchy. I'll think about how to fix it.

MikePopoloski, 2020-10-28:
Do you have an example of the double visit? I can't actually see how that would happen, because the statement trees are separated for each nested block.

MikePopoloski, 2020-10-28:
Aha, I was able to come up with a reproducer.

MikePopoloski, 2020-10-28:
Should be fixed now. Let me know if you still have issues.

closed 2020-10-28 by MikePopoloski