MikePopoloski/slang #331: Port declaration in module

Winging, 2020-10-29:
Hi Mike. I tested some cases I have which are supported by vcs. And found some are not supported by slang. Examples show below:
1.The port direction declaration shows first and data type declration shows in the end. 
```
module top(a. b);
       input a;
       var integer c;
       initial begin
               c = a; 
       end
       integer a;
```
2.The port direction declaration and data type declration shows after initial block. 
```
module top(a. b);
       var integer c;
       initial begin
               c = a; 
       end
       input integer a;
```
 After walking through the code,  I think these could be supported by judging if it is a port when seach name in scope. 
The code shows as below.  I am sure if there is a better way to support this since I am not quite familiar with the architeture of slang.
```
void Lookup::unqualifiedImpl(const Scope& scope, string_view name, LookupLocation location,
                             optional<SourceRange> sourceRange, bitmask<LookupFlags> flags,
                             SymbolIndex outOfBlockIndex, LookupResult& result) {
    // Try a simple name lookup to see if we find anything.
    auto& nameMap = scope.getNameMap();
    const Symbol* symbol = nullptr;
    if (auto it = nameMap.find(name); it != nameMap.end()) {
        for (auto& member : scope.members()) {
            if (member.name == name && memeber->kind == SymbolKind::Port) {
                flags = flags | LookupFlags::AllowDeclaredAfter;
            }
        }
.........
```

MikePopoloski, 2020-10-29:
Is the `a. b` in the port list a mistake? I'm not sure what that's supposed to do, no simulator I tried would accept it.

Assuming that it is a mistake and you meant to just put `a` there, I agree that #1 should work. I think the fix is actually when creating the internal port symbol, to merge the earliest location.

I believe that #2 is just invalid. Port list names are in their own namespace, inaccessible to the module itself. Other tools I've tried throw errors on it as well -- VCS is just bad at enforcing declare-before-use rules.

As a side note, non-ANSI style ports are not well supported by slang right now; you can see many TODOs in PortSymbols.cpp where things are unfinished.

Winging, 2020-10-29:
Sorry, my mistake. The handwritten example I showed is wrong , Following is the right case. I also put it on edaplayground
https://www.edaplayground.com/x/R4qw . It goes well on vcs and other commercial tools.
:
```
module top1(a);
       input a;
       var integer c;
       initial begin
               c = a; 
       end
       integer a;
endmodule

module top2(a);
       input a;
       var integer c;
       initial begin
               c = a; 
       end
       integer a;
endmodule

```

Winging, 2020-10-29:
Hmm, I tested the second example on cadence and mentor, it goes bad indeed.

MikePopoloski, 2020-10-30:
#1 should be fixed in 758e11ee79

closed 2020-10-30 by MikePopoloski