MikePopoloski/slang #332: Ability to make "UnknownModule" error suppressible and include those instances in --ast-json output

pistoletpierre, 2020-11-01:
I'm looking for a quick way to read in a design and get a complete list of instances in it. slang's --ast-json output + a bit of Python has gotten me 90% of the way there. The remaining 10% is dealing with undefined module instantiations. With what I'm doing it's more common than not to see instantiations of modules in a design for which HDL module definitions are either not available or deliberately excluded from a list of source files to read in and elaborate (standard cells, memories, other analog macros, submodules that have already been synthesized, etc...). One option is two make 2 passes: the first to get the list of errors and generate empty module definitions and the second to include the missing modules' generated empty definitions, The other is to add the ability to slang to treat undefined modules as black boxes, which seems like a good thing for it to be able to do.

I had a quick go at picking some potentially low-hanging fruit in [slang/source/symbols/InstanceSymbols.cpp](https://github.com/MikePopoloski/slang/blob/c0cf9a643f05df63268ecbfe561e9af37c9a62a8/source/symbols/InstanceSymbols.cpp#L221), but quickly realized that I was going to have to go down a rabbit hole to hack something together that would compile.

How long do you reckon it would take to become familiar enough with slang's structure to accomplish such a thing?

MikePopoloski, 2020-11-01:
Is the problem that you still want to see port connection information for the unknown modules in the AST? If not, it seems fairly easy to just suppress the UnknownModule error itself based on some option in the compilation options.

pistoletpierre, 2020-11-01:
I'm not even interested in the port connection information - just for an "InstanceBody"' to appear in the AST JSON output with "name" and "definition" fields. The "members" field of a missing module's "InstanceBody" could be empty for those instances. Right now if a particular module is missing, the AST JSON written to stdout leaves out any mention of any instances of that module. Compare the following 2 outputs with top.sv being the SV code block that follows the bash code block:

```bash
slang  -D MISSING_MODULE2 --ast-json -  top.sv | grep module2
#vs
slang                     --ast-json -  top.sv | grep module2
```

```systemverilog
// top.sv

`ifndef MISSING_MODULE2
module my_module2 (
  input  logic i_clk,
  output logic o_clk
);
endmodule
`endif

module my_module (
  input  logic i_clk,
  output logic o_clk
);

  my_module2 my_module2_inst (
    .i_clk(i_clk),
    .o_clk(o_clk)
  );

endmodule

module my_module3 (
  input  logic i_clk,
  output logic o_clk
);
  my_module my_module_inst (
    .i_clk(i_clk),
    .o_clk(o_clk)
  );
endmodule

module top (
  input logic i_clk,
  output logic [15:0] o_clks
);

  my_module3 my_module3_inst (
    .i_clk(i_clk),
    .o_clk(o_clks[15])
  );

  genvar i;
  generate
    for (i=0; i < 2; i++) begin : generate_test
      my_module my_module_inst (
        .i_clk(i_clk),
        .o_clk(o_clks[i])
      );
    end : generate_test
    for (i=0; i < 2; i++) begin : generate_test2
      my_module2 my_module2_inst (
        .i_clk(i_clk),
        .o_clk(o_clks[i])
      );
    end : generate_test2
  endgenerate

endmodule
```

To give a concrete example of what I'm going for: this Python script gets me all the instance paths in the `top` module (I'm sure it'll need to be added to in order to work with designs that use more SV language features but this is sufficient to get all the instances of that particular test case):

```python
#test.py
import json
import sys

def get_instances(ast_item, parent_hierarchy=""):
    result = []
    if ast_item["kind"] == "Root":
        for i, member in enumerate(ast_item["members"]):
            result += get_instances(member, parent_hierarchy)
    elif ast_item["kind"] == "CompilationUnit":   return result
    elif ast_item["kind"] == "Port":              return result
    elif ast_item["kind"] == "Net":               return result
    elif ast_item["kind"] == "Variable":          return result
    elif ast_item["kind"] == "Genvar":            return result
    elif ast_item["kind"] == "GenerateBlockArray":
        parent_hierarchy += f'.{ast_item["name"]}'
        for i, member in enumerate(ast_item["members"]):
            result += get_instances(member, parent_hierarchy=f'{parent_hierarchy}')
    elif ast_item["kind"] == "GenerateBlock":
        for i, member in enumerate(ast_item["members"]):
            if i == 0:
                if member["kind"] != "Parameter": # Setting a trap in case my assumption of this aspect of the ast dict structure isn't always the case
                    raise Exception(f"First member of GenerateBlock is not a parameter: revisit elif GenerateBlock clause.")
                current_param_val = f'{member["value"]}'
            else:
                result += get_instances(member, parent_hierarchy=f'{parent_hierarchy}[{current_param_val}]')
    elif ast_item["kind"] == "Instance":
        parent_hierarchy = f'{parent_hierarchy}.{ast_item["name"]}'
        result += [parent_hierarchy]
        result += get_instances(ast_item['body'], parent_hierarchy=f'{parent_hierarchy}')
    elif ast_item["kind"] == "InstanceBody":
        for i, body_item in enumerate(ast_item["members"]):
            result += get_instances(body_item, parent_hierarchy=f'{parent_hierarchy}')
    else:
        raise Exception(f"Unknown ast_item kind: {ast_item['kind']}")
    return result

ast = json.loads(open(sys.argv[1], "r").read())
insts = get_instances(ast)
print(json.dumps(insts, indent=4))
```

```bash
slang  -q --ast-json - top.sv > ast.json
python test.py ast.json
```

```python
# Output:
[
    ".top",
    ".top.my_module3_inst",
    ".top.my_module3_inst.my_module_inst",
    ".top.my_module3_inst.my_module_inst.my_module2_inst",
    ".top.generate_test[0].my_module_inst",
    ".top.generate_test[0].my_module_inst.my_module2_inst",
    ".top.generate_test[1].my_module_inst",
    ".top.generate_test[1].my_module_inst.my_module2_inst",
    ".top.generate_test2[0].my_module2_inst",
    ".top.generate_test2[1].my_module2_inst"
]
```

MikePopoloski, 2020-11-02:
I think the easiest solution might be to have a new kind of symbol, UnknownInstanceSymbol or even just UnknownSymbol that can be inserted as a placeholder (and can hold a pointer to the syntax node). Trying to reuse an InstanceSymbol will require a lot of otherwise unnecessary modifications to satisfy all of its current requirements. You can already use the DiagnosticEngine::setSeverity method to mark the UnknownModule diagnostic as Ignored, though it's not exposed via a command line option to the driver tool. I'll think more about how applicable such a feature would be to other users.

MikePopoloski, 2020-11-28:
I added an UnknownModuleSymbol which should do what you want here.

closed 2020-11-28 by MikePopoloski