MikePopoloski/slang #335: Report a better error for variable definitions that refer to a module/interface definition

MikePopoloski, 2020-11-14:
It's a common error to leave off the parentheses in a module or interface instance and end up with something that gets parsed as a variable declaration instead.

thirtythreeforty, 2022-03-03:
For those attempting to Google error messages:
```
test_formal.sv:22:5: error: 'test_if' is an interface definition but is used like a type
    test_if i, j;
    ^~~~~~~
```
... which _is_ rather vexing :)

The correct instantiation is:
```systemverilog
    test_if i(), j();
```

MikePopoloski, 2022-07-16:
As of 8b15c58b1903313ad9906cc1dcc6108441a3db2e slang now reports an error like this:
```systemverilog
interface test_if;
endinterface

module m;
    test_if i, j;
endmodule
```

```
source.sv:5:14: error: interface instantiation is missing port list parentheses
    test_if i, j;
             ^
source.sv:5:17: error: interface instantiation is missing port list parentheses
    test_if i, j;
                ^
```

closed 2022-07-16 by MikePopoloski