MikePopoloski/slang #361: Support specifying command line options in a separate file

MikePopoloski, 2020-11-14:


krupan, 2021-03-26:
Is the psuedo-standard "dot f file" that most simulators support?  If so this is something I would like to help with.

MikePopoloski, 2021-03-26:
Yes, that's right. Patches are definitely welcome.

martinda, 2021-09-01:
Some circles call them "manifest files".

MikePopoloski, 2021-09-01:
Some notes on this since my original comment is conspicuously blank:
* The file should ideally accept basically any command line option, not just file lists
* File should support comments and possibly some primitive form of include directive
* This is most useful when paired with #360 so that entries in the file would be portable from other tools

MikePopoloski, 2022-01-30:
Support for -f and -F command files have been added.

closed 2022-01-30 by MikePopoloski