MikePopoloski/slang #376: error in JSON generated for DPI-C declaration

jankcorn, 2020-12-30:
The following verilog:
module VsimTop ( input wire CLK, input wire nRST);
    import "DPI-C" function void dpi_msgSend_enq(input int enq);
    always @(posedge CLK) begin
            dpi_msgSend_enq(32'h0);
    end
endmodule
Produces a json fragment:
          {
            "name": "dpi_msgSend_enq",
            "kind": "Subroutine",
...
            "arguments": [
              "lifetime": "Automatic",
              "isConstant": false,
              "isCompilerGenerated": false,
              "direction": "In"
            ]
          }
which fails to parse using the cJSON library.  

The interior of the array for "arguments", must be 'elements', not 'members':
     https://www.crockford.com/mckeeman.html
(you need to put '{'/'}' around the members)

MikePopoloski, 2020-12-31:
Fixed in 407a4fad07

closed 2020-12-31 by MikePopoloski

jankcorn, 2020-12-31:
So fast!
Thanks very much