MikePopoloski/slang #377: problem in hierarchical names/interface reference?

jankcorn, 2020-12-31:
The following verilog:
interface PipeOut#(width = 32);
    logic deq__ENA;
    modport server (input  deq__ENA);
    modport client (output deq__ENA);
endinterface
module AxiTop ();
    PipeOut#(.width(32)) __traceMemory$out();
endmodule
module ZynqTop ();
    wire enable;
    AxiTop test ();
    assign test.__traceMemory$out.deq__ENA = enable;
endmodule

Produces the following error:
internal compiler error: slang/source/binding/Expression.cpp:60: Default case should be unreachable!

Note: When the hierarchical name ref is to a 'wire', no problem.

MikePopoloski, 2021-01-02:
This does not crash for me. Are you sure that's the exact code you're having a problem with?

jankcorn, 2021-01-02:
I did a fresh pull from your repo and it works fine.  I apologize!  
It must have been a side effect of some local edits.
Thank you

closed 2021-01-02 by jankcorn