MikePopoloski/slang #380: Question: Is it possible to ignore certain errors?

zegervdv, 2021-01-14:
I'm trying to use slang as a linter to find issues with my code faster.
I can run it on a single file to quickly see issues, but it flags the included modules as unknown.

Is there a way to ignore those warnings? Maybe through some `--lint-only` flag?

MikePopoloski, 2021-01-14:
There is an `--ignore-unknown-modules` flag to not report errors about unknown modules. You could probably also pass each file via `-v` to mark them as  "library" files, so they don't automatically instantiate modules as top level. I could see adding a `--lint-only` flag or similar that would not try to instantiate any modules at all automatically.

zegervdv, 2021-01-14:
I'm not seeing the `--ignore-unknown-modules` in the help. 
And slang also does not recognize it:
```
slang: unknown command line argument '--ignore-unknown-modules'
```
I am on the 0.6 release version

zegervdv, 2021-01-14:
Although it reports itself as `slang version 0.5.0daf9da`

MikePopoloski, 2021-01-14:
Oh hmm, did you download this from the release? Or build it from source?

zegervdv, 2021-01-14:
I got it from the releases

MikePopoloski, 2021-01-14:
OK, the CI build -> release process must be broken somehow and attaching old binaries. In the mean time you should build from source.

zegervdv, 2021-01-14:
Alright I'll give it a try

zegervdv, 2021-01-14:
I keep running into build issues on the latest hash:
```
slang/source/binding/OperatorExpressions.cpp: In static member function ‘static slang::Expression& slang::ConditionalExpression::fromSyntax(slang::Compilation&, const slang::ConditionalExpressionSyntax&, const slang::BindContext&, const slang::Type*)’:
slang/source/binding/OperatorExpressions.cpp:954:53: error: no matching function for call to ‘slang::BindContext::setAttributes(slang::ConditionalExpression&, const slang::SyntaxList<slang::AttributeInstanceSyntax>&) const’
  954 |     context.setAttributes(*result, syntax.attributes);
```

MikePopoloski, 2021-01-14:
What compiler are you using? The CI builds are running fine on HEAD.

zegervdv, 2021-01-15:
I’m on Ubuntu 20.04 with gcc-9.

zegervdv, 2021-01-15:
Had to upgrade to gcc-10.

I am seeing the `--ignore-unknown-modules` option, but the output is still the same. It is still reporting errors for the unknown modules.

MikePopoloski, 2021-01-15:
Can you post an example that reproduces the issue?

zegervdv, 2021-01-15:
Upon further examination I noticed the errors about the modules have disappeared, but I am still seeing them on packages and interfaces.

This is my minified test:
``` systemverilog
module foo #() (
  input logic clk,
  input logic reset,
  itf.bus command
);

  bar #(.param(4)) test_i (.clk(clk));
endmodule;
```

Which returns:
```
$ slang --ignore-unknown-modules test.sv
test.sv:4:3: error: unknown interface 'itf'
  itf.bus command
  ^~~
warning: no top-level modules found in design [-Wno-top]

Build failed: 1 error, 1 warning
```

zegervdv, 2021-01-15:
Even without the `--ignore-unknown-modules` flag slang is not reporting the unknown bar module when there are interfaces.

Compare with v0.5:
```
$test.sv:4:3: error: unknown interface 'itf'
  itf.bus command
  ^~~
test.sv:7:3: error: unknown module 'bar'
  bar #(.param(4)) test_i (.clk(clk));
  ^~~
warning: no top-level modules found in design [-Wno-top]

Build failed: 2 errors, 1 warning
```

MikePopoloski, 2021-01-15:
v0.6 made unknown modules not error if they are in an otherwise uninstantiated module. Modules with interface ports are not automatically instantiated (per language rules).

There's no way to suppress errors about missing interface in ports. I suppose one could be added; the use case seems a little weird to me because you won't be able to do type checking or name resolution or maybe even parameter evaluation for modules where you don't know the interface ports. Also you can only even know it's an interface port if you give it a modport restriction -- if it's just the interface name itself then it's totally ambiguous whether it's an interface or just a type name.

zegervdv, 2021-01-15:
I understand. A lot of the potential errors become invisible if you cannot see the type/interface of a port or signal.
I was trying to use slang a sort of static linting to see some errors sooner but of course that will always be limited when you can only see one file at a time.

Thanks for your time and work.

MikePopoloski, 2021-01-16:
You can pass `--parse-only` to only get parser errors. This doesn't do name resolution or type checking, so it's not as comprehensive but may be closer to what you want.

MikePopoloski, 2021-04-09:
I also added a `--lint-only` flag that wraps up some of the other options. It doesn't solve the interface port problem but as previously mentioned there's no way to solve that generally without just turning off all errors about unknown type names.

closed 2021-04-09 by MikePopoloski