MikePopoloski/slang #381: support for 'assert property'

jankcorn, 2021-01-17:
Do you have any plans for supporting property declarations and assert?

As concrete examples, I ran into these:
wrapper.v:31:4: warning: language feature not yet supported [-Wnot-supported]
   chk_valid: assert property (@(posedge clk) check_valid == 1'b1);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
traffic_good.sva:33:36: error: expected expression
        yellow_first |=> red_first[*1:$] ##1 green_first[*1:$] ##1 yellow_first);
                                                  ^
traffic_good.sva:67:42: error: expected expression
         (waiting_first && waiting_main)[*MAX_WAIT]);
                                                               ^
Thanks very much for the parser!

MikePopoloski, 2021-01-17:
Yes of course, all language features will be supported eventually.

MikePopoloski, 2021-01-18:
I'm going to close this out. Tracking support for various language features can be done by checking this page: https://sv-lang.com/language-support.html and by checking release notes of new releases.

closed 2021-01-18 by MikePopoloski

jankcorn, 2021-01-18:
Thanks very much for pointing me back to the feature tracker!
This is exactly what I need.
BTW, I am trying to implement a simple slang->BTOR2 file output so I
can connect to the PONO model checker.

Thanks for all your help!

lzxqaq, 2023-11-24:
> Thanks very much for pointing me back to the feature tracker! This is exactly what I need. BTW, I am trying to implement a simple slang->BTOR2 file output so I can connect to the PONO model checker. Thanks for all your help!
> […](#)

Was this project successful?

jankcorn, 2023-11-25:
I was successful at parsing analyzing all the assert property variants
I found, but stopped before translating the resulting structures into
verilog FSMs (for model checking)