MikePopoloski/slang #384: Annotating data to AST or SyntaxTree?

cryptoplex, 2021-02-02:
I would like to use slang to do some SystemVerilog processing in C++. I need to have the ability to annotate data to the various nodes of the AST and/or the SyntaxTree.

Is there an intended way to do such annotation? I looked through the documentation, 'tests/unittests/VisitorTests.cpp', and some of the code, but I'm not finding anything.

Thanks,
Mike

MikePopoloski, 2021-02-02:
There isn't such a mechanism right now. You can always use a plain old hash table on the address of the node -- the only difference is that it would be a bit more efficient to have a field on the node itself if you are going to be storing data on many nodes, and a bit less efficient if the data is sparse.

cryptoplex, 2021-02-02:
Okay. Thanks. I think it would be a good future enhancement to add support for annotating user-defined data.

MikePopoloski, 2021-02-02:
Is there something you're looking for with an API change that you can't get with a hash lookup?

cryptoplex, 2021-02-03:
That will work in a general sense. But, if the goal of the project is to provide support for nontrivial SystemVerilog analysis and processing, annotating user-defined data to the ASTs is going to be a common use case. Requiring users to provide their own node-to-user-data hashing feels like a workaround for incomplete support (at least, to me). Also, if one is not careful, using pointer values in that way could become a source of nondeterminism.

MikePopoloski, 2021-02-07:
Ok, I will keep that in mind. I can't add an API without a concrete example of how it will be used, so I will close this out for now.

closed 2021-02-07 by MikePopoloski