MikePopoloski/slang #385: Easy way to get source/sink of a port symbol?

Kuree, 2021-02-26:
We're working on a tool that uses `slangcompiler` as a library to interactively analyze the design. One of the tasks is to figure out the connectivities of a given port.

I remembered that `PortSymbol` used to have access to its own connections, but it seems that the functionality got refactored to `InstanceSymbol`. What is the canonical way to compute the source/sink for a given `PortSymbol`?

Thanks.

PS:
Not sure if this should be another issue but it seems like `getRoot().lookupName()` returns variable symbol instead of port symbol, if the given path is a port. Is there an easy way to get port symbol instead?

MikePopoloski, 2021-02-26:
A PortSymbol on its own is not sufficient to get the connection; you also need to know which instance you want the connection for, because PortSymbols are shared by all instances that use the same InstanceBody. Assuming you have an instance, you can call instance.getPortConnection(portSymbol) to get the connection for that specific instance and port.

lookupName follows hierarchical lookup rules, which can't ever be used to look up a port, which exist in their own namespace accessible only by instantiations. Ports don't even need names, like with non-ansi concatenations, so there would be no way to find those by name even if you wanted to.

MikePopoloski, 2021-02-26:
Assuming you have an InstanceBody, you can find a port symbol using the findPort() method. Again, you can't find all possible ports that way but it's a convenience method.

Kuree, 2021-02-26:
> Assuming you have an instance, you can call instance.getPortConnection(portSymbol) to get the connection for that specific instance and port.

Thanks. I will try that.

> Assuming you have an InstanceBody, you can find a port symbol using the findPort() method. Again, you can't find all possible ports that way but it's a convenience method.

Yes I'm doing that right now.

closed 2021-02-26 by Kuree