MikePopoloski/slang #386: Segfault when using slangcompiler in Debug build

Kuree, 2021-02-27:
It seems like `SourceManager` will trigger a segfault when set to cmake debug mode, if the library is added as a sub directory. Here is the minimal example to reproduce.

`example.cc`:
```C++
#include "slang/text/SourceManager.h"

int main() {
    slang::SourceManager manager;
}
```

`CmakeLists.txt`:
```cmake
cmake_minimum_required(VERSION 3.12)
project(example)

set(CMAKE_CXX_STANDARD 17)
add_subdirectory(slang)

add_executable(example example.cc)
target_link_libraries(example PRIVATE slangcompiler)
target_include_directories(example PRIVATE slang/include)
```

How to reproduce:
```
# put example.cc and CMakeLists.txt in the current working directory
git clone https://github.com/MikePopoloski/slang/
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
make -j
```

Then run the binary `example`
```
$ ./example 
Segmentation fault (core dumped)
```

Tested with `ubuntu:latest`, `ubuntu:devel` docker images. All build tools are installed via `apt`. It seems to at least reproducible with the following compilers:
- gcc-9
- gcc-10

MikePopoloski, 2021-02-27:
The problem is that the slang build defines the DEBUG macro when it's a debug cmake build but your example.cc translation unit does not define it, so when it includes SourceManager.h (and other headers) there is a mismatch in the code generated.

So short term make sure you define DEBUG for debug builds in your cmake project. Longer term I should spend more time thinking about best practices for using slang via cmake as a library.

Kuree, 2021-02-27:
Thanks. Defining `DEBUG` macro fix the issue.

closed 2021-02-27 by Kuree