MikePopoloski/slang #388: Support for older linux versions?

DeeeeLAN, 2021-03-04:
How hard would it be to add support for older versions of Linux. At work we are running CentOS 7 which has an older version of libc, so the binaries don't run:
```
./slang: /lib64/libm.so.6: version `GLIBC_2.27' not found (required by ./slang)
./slang: /lib64/libm.so.6: version `GLIBC_2.29' not found (required by ./slang)
./slang: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by ./slang)
./slang: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by ./slang)
./slang: /lib64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./slang)
./slang: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./slang)
./slang: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./slang)
```
I tried building from source but the newest version of gcc currently installed is 7.4.0. If it was possible to support an older version of gcc it would help similarly. 

I understand sometimes this type of request is not possible.

MikePopoloski, 2021-03-05:
Unfortunately the old versions of gcc are lacking a lot of modern C++ support, and the libraries are missing too (such as the filesystem std library that slang uses). I don't really have the time and resources to support many build configurations, as this is just an open source project that I work on in my spare time.

DeeeeLAN, 2021-03-09:
Understood, thanks!

closed 2021-03-09 by DeeeeLAN

jiao-x, 2025-02-10:
你可以使用一个支持slang的linux系统对slang进行编译，编译的时候把动态链接库也一起编译进去，编译好后就可以直接拿到旧版本的linux系统上使用了