MikePopoloski/slang #393: parsing problem in property_expr

jankcorn, 2021-05-13:
First, thanks for the parser support for property/sequence exprs!

The statement:
     assert property (@(posedge dif_clk) ((dif_req) and dif_req) ##0 dif_gnt2);
gives a syntax error.  
I believe that it _might_ be caused by the fact that both
    '( property_expr )' and '( sequence_expr )' are valid token strings 
(the error goes away if the parens around '(dif_req)' are deleted. 

A

MikePopoloski, 2021-05-13:
Yeah, unfortunately you can't know if a parenthesized expression is a normal, sequence, or property expression without infinite lookahead so I do a fixup approach, but it's clearly still buggy. I'll look at it more.

MikePopoloski, 2021-05-15:
Fixed in cbf7ea8966841866dc7a7ae3fad51e538f86cd22

closed 2021-05-15 by MikePopoloski

jankcorn, 2021-05-15:
Thanks very much!