MikePopoloski/slang #394: question about JSON for 'max' in 'SequenceConcat'

jankcorn, 2021-05-13:
For now, it appears that you generate a 'max' value of 0 for '$' in 
'cycle_delay_const_range_expression' items, e.g.:
    ##[0:$]
Unfortunately, it makes it hard to distinguish this from the usage:
    ##[0:0]

Could you possibly add a flag in the JSON for the case when 'max' == '$'?
Thanks!

MikePopoloski, 2021-05-13:
Do you have an example that causes it? When I try with the following:
```
module m;
    int a;
    assert property (a ##[0:$] a);
endmodule
```
I see this output, which matches what I'd expect:
```
                   {
                    "sequence": {
                      "kind": "Simple",
                      "expr": {
                        "kind": "NamedValue",
                        "type": "int",
                        "symbol": "1643202843880 a"
                      }
                    },
                    "min": 0,
                    "max": "$"
                  }

jankcorn, 2021-05-14:
My error, sorry!

I looked at a few examples and (mistakenly) extrapolated that the
datatype for 'max' was int.

Sorry for the confusion and thanks for the clarification!

closed 2021-05-15 by MikePopoloski