MikePopoloski/slang #395: ambiguity in JSON for property_expr with implication

jankcorn, 2021-05-13:
It appears that an identical JSON is generated for:
    a |-> b
and
    a |=> b
Do you eventually intend to keep these distinct in the JSON or are you planning to put the LRM rewrite rule 'F.3.4.3.3 Derived nonoverlapping implication operator' into the parser?
 For example:
       a |=> b
becomes
       a ##1  1 |-> b

Thanks!

MikePopoloski, 2021-05-14:
I'll add an enum. I'm only planning to do the parts of the rewrite algorithm that are needed to expand out sequence and property instances and to enforce all semantic rules. The rest seems like it should be deferred until simulation.

jankcorn, 2021-05-14:
Yes, I was planning on doing the rewrite in my code (downstream from slang).

Thanks for adding the enum!

MikePopoloski, 2021-05-15:
Fixed in 4981a18f

closed 2021-05-15 by MikePopoloski