MikePopoloski/slang #397: Slang failed to build as shared library on Windows (MSVC)

gasrodriguez, 2021-06-14:
Hi! I'm trying to build slang as a shared library on windows and I'm having some link issues. 

Here's the log and the error and warnings:
[slang-shared-log.txt](https://github.com/MikePopoloski/slang/files/6651583/slang-shared-log.txt)
[slang-shared-errors.txt](https://github.com/MikePopoloski/slang/files/6651582/slang-shared-errors.txt)

I can build the slang.exe executable successfully with the default configuration, and the only difference between the two configurations is the BUILD_SHARED_LIBS set to true ("x64-Release" works ok, "x64-Release-Shared" fails):
```json
{
	"configurations": [
		{
			"name": "x64-Release",
			"generator": "Ninja",
			"configurationType": "Release",
			"inheritEnvironments": [ "msvc_x64_x64" ],
			"buildRoot": "${projectDir}\\out\\build\\${name}",
			"installRoot": "${projectDir}\\out\\install\\${name}",
			"cmakeCommandArgs": "",
			"buildCommandArgs": "",
			"ctestCommandArgs": "",
			"variables": [
				{
					"name": "BUILD_SHARED_LIBS",
					"value": "False",
					"type": "BOOL"
				}
			]
		},
		{
			"name": "x64-Release-Shared",
			"generator": "Ninja",
			"configurationType": "Release",
			"buildRoot": "${projectDir}\\out\\build\\${name}",
			"installRoot": "${projectDir}\\out\\install\\${name}",
			"cmakeCommandArgs": "",
			"buildCommandArgs": "",
			"ctestCommandArgs": "",
			"inheritEnvironments": [ "msvc_x64_x64" ],
			"variables": [
				{
					"name": "BUILD_SHARED_LIBS",
					"value": "True",
					"type": "BOOL"
				}
			]
		}
	]
}
```
I'm using Microsoft Visual Studio Community 2019 Version 16.10.0 with a current version of CMake. Is there anything I can try? Thanks!

gasrodriguez, 2021-06-14:
I also tried enabling shared library and disabling static:
```json
				{
					"name": "SLANG_MSVC_STATIC_RUNTIME",
					"value": "False",
					"type": "BOOL"
				}
```
and the build process continued a bit further, but eventually fails nonetheless
[slang-shared-log.txt](https://github.com/MikePopoloski/slang/files/6651607/slang-shared-log.txt)

But it does create the slangcore.dll file in:
...\slang\out\build\x64-Release-Shared\bin\slangcore.dll
Does it mean that the shared lib was created successfully this time?
GR.

MikePopoloski, 2021-06-16:
On Windows, symbols in DLLs are not automatically exported. CMake has a feature to automatically export everything that yo ucan try: https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/

Note that usually a DLL is designed with a C API so that you don't have to worry about ABI concerns. You can probably make it work as a shared lib exporting the full C++ API that slang provides but you probably have to be careful about matching the ABI with your own application.

gasrodriguez, 2021-06-16:
Hi Mike, thanks for the reply. Ok so is safe to say that building slang as shared lib is not currently supported for windows, right? I may try to export all symbols but as you say, exposing a C api would be preferred (since ultimately I'd like to create a binding for Go, and C is supported out-of-the-box; for C++ something like SWIG is required). For the main classes should be straightforward, but I'm not sure if/how the wrapping of the SyntaxNode derived classes would work (without touching the Python scripts). Also, I'd like to be able to "derive" from SyntaxVisitor and SyntaxRewriter or at least set custom callbacks from Go, but the C wrapper may make that not possible (?
Sorry if these are dummy questions and perhaps out of topic for this issue. Maybe you didn't think in creating bindings for others languages and that's a feature currently not supported, so please feel free to mark the issue as Closed if that's the case :)

MikePopoloski, 2021-06-16:
Yes, right now there is no support for a C API. I'd certainly like to have one but it's definitely not a small amount of work to build and maintain so I haven't had time. If you wanted to contribute a C API it would be welcome, but it will require some thought how best to wrap certain cases, as you noted.

MikePopoloski, 2021-08-14:
I'm going to close this out; the real ask here is for a C API which I don't have time to work on right now but can be contributed if someone wants to.

closed 2021-08-14 by MikePopoloski