MikePopoloski/slang #398: Parsing interface port assignment to JSON fails

YakirMishli, 2021-06-17:
```
interface my_ifc;

wire [7:0] data;

modport master 
(
    output data,
);
endinterface

module test_marcos     
    (
        input  var  logic                  clk,        
        my_ifc                             my_if,
        my_ifc.master                      my_if_mod,                   
        output reg                         data_out_2
    );
    
    always_comb begin
        my_if_mod.data = 7'd0;
    end        
    

endmodule
```

the JSON result not detecting the interface port assignment:

```
{
            "name": "",
            "kind": "ProceduralBlock",
            "addr": 1578880722584,
            "procedureKind": "AlwaysComb",
            "body": {
              "kind": "Invalid",
              "child": {
                "kind": "Block",
                "blockKind": "Sequential",
                "body": {
                  "kind": "List",
                  "list": [
                    {
                      "kind": "Invalid",
                      "child": {
                        "kind": "ExpressionStatement",
                        "expr": {
                          "kind": "Invalid",
                          "type": "**<error>**",
                          "child": {
                            "kind": "Assignment",
                            "type": "**<error>**",
                            "left": {
                              "kind": "Invalid",
                              "type": "**<error>**"
                            },
                            "right": {
                              "kind": "IntegerLiteral",
                              "type": "bit[6:0]",
                              "value": "7'b0",
                              "constant": "7'b0"
                            },
                            "isNonBlocking": false
                          }
                        }
                      }
                    }
                  ]
                }
              }

```

MikePopoloski, 2021-06-17:
It is not generally possible to elaborate (and/or do type checking) for modules with interface ports that aren't connected to something. Consider that given parameters in the interface, members can have literally any type, and so there's no way to say what type an expression like `my_if_mod.data = 7'd0;` will have until my_if_mod is connected to a real instance.

In your example here it would work ok because the interface doesn't have any parameters and all types are known concretely. I could imagine an improvement to slang would be to make a fake instance for all known interface ports so that you get a bit more info for modules like this that aren't actually instantiated. Note that this falls apart if you use the `interface.master` syntax to declare your interface ports, as then you really don't know anything about the port until you connect it.

YakirMishli, 2021-06-18:
Thanks for answer, it make sense.
I use slang to detect dangling ports in a module so I compiled only the specific module.
After your explanation, I will compile the whole project to JSON and it will solve my problem.

Your application is great! I hope to contribute in future.

MikePopoloski, 2022-01-24:
Closing this issue since I don't think there's anything else to do here.

closed 2022-01-24 by MikePopoloski