MikePopoloski/slang #399: Fail to build on non-x86 platforms

jesec, 2021-07-03:
https://github.com/MikePopoloski/slang/blob/master/source/numeric/SVIntHelpers.h:
```
source/numeric/SVIntHelpers.h:157:17: error: '_addcarry_u64' was not declared in this scope
  157 |         carry = _addcarry_u64(carry, src[i], value, &result);
      |                 ^~~~~~~~~~~~~
```

Apparently, Slang uses x86 intrinsic. Is it possible to replace them with pure C++ for better portability?

MikePopoloski, 2021-07-04:
Yes, it's possible. What platform are you building for?

jesec, 2021-07-04:
arm64, but we may use riscv in the future.

MikePopoloski, 2021-07-05:
I added a plain C implementation for non-x86 platforms. I have no way to test this though, or ensure that other things in the codebase don't fail to build. Feel free to put up pull requests if you run into other issues.

closed 2021-07-05 by MikePopoloski

jesec, 2021-07-06:
> I added a plain C implementation for non-x86 platforms. I have no way to test this though, or ensure that other things in the codebase don't fail to build. Feel free to put up pull requests if you run into other issues.

👍 Thank you!