MikePopoloski/slang #400: Documentation on Structure of AST

saw235, 2021-08-09:
I am trying to figure out how to consume the AST that is generated. Is there some documentation on how the structure is organized? 

Is there also a way to include the source information on the ast-json output?

MikePopoloski, 2021-08-10:
You're referring to the AST JSON? The output is the elaborated design tree. You can consume it via any tool that can read and manipulate JSON files. If you want to know exactly which fields can appear in the output you can look at the serializeTo methods in the source code; there isn't documentation on specific fields for each AST class. 

In what way would you like source information to be in the output?

saw235, 2021-08-10:
For example, if a downstream tool uses the AST to elaborate the RTL into a Data Flow Graph, then it would be convenient to know where the issue occurred if the semantics is incorrect. Such as the line information. 

I also noticed that there's the address field, but I'm not sure I understand what is it for.

MikePopoloski, 2021-08-10:
The address is simply an identification mechanism, since names are not unique. Two references to a symbol will point to the same address.

The JSON output isn't really meant as a stable and efficient interface to a full-fledged tool. I could put file and offset into each element but it would make the output comically large (it's already very verbose). Still, if you want it I can look at adding an option.

Linking as a library is the intended way to use slang as a frontend, since that will give you complete information and an API for interacting with it. If your project isn't in C++ though this is currently difficult, as I haven't had time to write a C or foreign language interface. Definitely an area that could be improved through contributions though.

MikePopoloski, 2021-08-13:
I'm going to close this; if you have any specific questions feel free to ask.

closed 2021-08-13 by MikePopoloski