MikePopoloski/slang #401: [GCC9] AssertionExpr.cpp:217:12: error: enumeration value ‘DisableIff’ not handled in switch

ajelinski, 2021-08-12:
The latest commits break building slang with GCC 9.3.0:
```
  /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/slang_1628752251297/work/source/binding/AssertionExpr.cpp: In member function ‘void slang::AssertionExpr::requireSequence(const slang::BindContext&, slang::DiagCode) const’:
  /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/slang_1628752251297/work/source/binding/AssertionExpr.cpp:217:12: error: enumeration value ‘DisableIff’ not handled in switch [-Werror=switch]
    217 |     switch (kind) {
        |            ^
  cc1plus: all warnings being treated as errors
  make[2]: *** [source/CMakeFiles/slangcompiler.dir/build.make:82: source/CMakeFiles/slangcompiler.dir/binding/AssertionExpr.cpp.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:252: source/CMakeFiles/slangcompiler.dir/all] Error 2
  make: *** [Makefile:149: all] Error 2
```

The last commit successfully used to build slang is yesterday's https://github.com/MikePopoloski/slang/commit/ccb99aa7d0162496b20af91ac1d542612618b0fb.

MikePopoloski, 2021-08-12:
Fixed in 4167c73

closed 2021-08-12 by MikePopoloski

ajelinski, 2021-08-13:
Fantastic! Thanks.