MikePopoloski/slang #402: Project name conflict with S-Lang Library

jrudess, 2021-08-13:
I'm packaging this project for Arch Linux (aur repository) and ran into some minor issues because another package is already using the same name (https://www.jedsoft.org/slang/).

So I thought I'd check with you to see how tied you are into the name "slang".  I'm changing the name of the binaries and shared-libraries to use slang-verilog right now.

MikePopoloski, 2021-08-13:
Hmm, that's unfortunate. I don't think slang is the greatest name in the world or anything like that but changing all occurrences would be a decent amount of work at this point, so I'd prefer not to unless it became critical to do so.

jrudess, 2021-08-13:
I can't think of any critical reason to do so, but just thought I'd check before finalizing the patch set needed for the build and install modifications.

The only other change I've identified for packaging is to remove the fmt/xxhash/catch2 code in the external directory and build with the preexisting packages for these libraries, but my C++/CMake is very stale, so I'm still working through some compilation issues when doing that.

closed 2021-08-13 by jrudess

MikePopoloski, 2021-08-13:
Ok, let me know if you need help. The dependencies are copied into the repo to avoid the headaches associated with external packages, but I assume that's not allowed for the distro you're targeting?

jrudess, 2021-08-14:
The 0.7 release had some issues related to the fmt API and some deprecated functions, but these are resolved in HEAD.  If it's not an inconvenient time, would you be able to push out a new release?

If you're interested, [here](https://gist.github.com/jrudess/ae704a0e1ca139c9b3942420595ce9b7) is the patch file I'm using.  It mostly modifies just cmake files, but I had to tweak one xxhash #include due to an install-location difference for Arch.

And the cmake command:
```
cmake -DCMAKE_CXX_COMPILER=g++ -DCMAKE_BUILD_TYPE=Release -DxxHash_DIR=$srcdir/..  -DCMAKE_INSTALL_PREFIX=/usr -DSLANG_USE_SYSTEM_LIBS=True -DBUILD_SHARED_LIBS=True ..
```

MikePopoloski, 2021-08-14:
Yes, good timing actually. I've been finishing up this release and was able to push it out this morning.