MikePopoloski/slang #428: interface port connected with hierarchy interface instance reference has issue

aegean0228, 2021-09-09:
module Top();
    dut inst_0(inst1.intf_inst);
    sub inst1();
endmodule
 
module dut(interface pi);
endmodule

module sub();
    intf intf_inst();
endmodule

interface intf();
endinterface

it should be a valid case, but i saw error

MikePopoloski, 2021-09-09:
I have not yet added support for the generic interface port keyword. If you explicitly name the interface type expected for the port then this works fine.

aegean0228, 2021-09-09:
change the case :
module Top();
    dut inst_0(inst1.intf_inst_1);
    dut inst_1(inst2.intf_inst_2);
    sub_1  inst1();
    sub_2  inst2();
endmodule

module dut(intf pi);
    parameter P = pi.P;
endmodule

module sub_1();
intf #(0)intf_inst_1();
endmodule

module sub_2();
intf #(1)intf_inst_2();
endmodule

interface intf();
parameter P = 0;
endinterface

if you check the parameter Top.inst_1.P, the value is 0 instead of 1 which is wrong.
both instance Top.inst_0 and Top.inst_1 share the same instanceBody which is wrong as the port connection pointing to different 
interface instance. 
from my debugging , findInterfaceInstanceKeys doesn't work properly as sub_1 inst(); is not elaborated yet.
could you please check it ?

MikePopoloski, 2021-09-10:
You're right, this is a bug. Honestly though I would have thought to fix it by outright disallowing hierarchical references to connect interface ports -- this seems like a really terrible feature and it's not clear from the LRM that it's actually allowed. The LRM tries very hard to make it clear that hierarchical references are never allowed in parameters and constant expressions, but if hierarchical references to interface ports are allowed you can essentially reference any parameter in the design hierarchically.

aegean0228, 2021-09-11:
yes, actually VCS allow hierarchical reference to parameter in any compile time constant context.
LRM is not clear about it, but unfortunately from my experience it's widely used by customers.
I am wondering how slang can accomplish it as it only top-down resolve the design once

MikePopoloski, 2021-09-11:
On the subject of hierarchical references in parameters and constant contexts, the LRM is extremely clear.

6.20.2 Value parameters:
> Hierarchical names are not allowed

13.4.3 Constant functions:
> - Constant functions shall contain no hierarchical references
> - All parameter values used within the function shall be defined before the use of the invoking
constant function call (i.e., any parameter use in the evaluation of a constant function call constitutes
a use of that parameter at the site of the original constant function call).
> - If constant functions use any parameter value that is affected directly or indirectly by a defparam
statement (see 23.10.1), the result shall be undefined. This can produce an error or the constant
function can return an indeterminate value.

The syntax allowed for interface port connections is unclear, they say almost nothing about it unfortunately. I think the intent is pretty clear though, that the SystemVerilog language disallows hierarchical names in all compile-time contexts. It's worth considering support for this anyway for compatibility purposes with VCS (presumably they added it because customers wanted it) but we should be clear that such a feature is not actually allowed by SystemVerilog.

aegean0228, 2021-09-13:
yes, there are quite a lot of features asked by customers which are not allowed by LRM.
we have a small team trying to build a systemverilog simulator completing with other commercial tools.
So support features other tool supported is important for us, that's the reason why i ask it.
we are looking around to see if there are open source projects which can be part of our frontend,
i think overall slang did great job, but based on my research those days, i did not figure out way that based on slang current infrastructure, this feature can be supported.  we used to build an elaborated tree during elaboration , and we do elaboration on the tree first, and after everything is finalized , then we will eventually expand the design.
i think slang has a very different elaboration flow compared those tools i worked before, so just wondering if there is possiblity support it without change current  infrastructure

MikePopoloski, 2021-09-13:
I'd have to think about it more but it's probably doable. slang's model is very flexible, in that symbols and scopes are lazily elaborated as needed, which means a lot of ordering constraints are simply not a factor. Right now parameters are all forcibly elaborated whenever a scope is first entered but it's possible that could be changed to also be lazily done.

jrudess, 2021-09-17:
I've also run into cases where I need hierarchical references to module parameters.   This commonly happens for testbench logic on heavily parameterized designs.  Parameterized interfaces are horrible for a testbenche due to the 'viral' nature of parameters. 
 I.e. every class that instantiates that virtual interface must now be parameterized, and every class that instantiates that class, all the way to your top level UVM test class.

To avoid needing a massive number of parameters for every class, we size signals in testbench interfaces to the maximum width we plan to ever support.   This creates a width mismatch between the RTL and the interface which causes new issues.
* Wildcard .* connections are not allowed when the signals have mismatching widths
* Manual port connections are allowed, but cause a compile-time width mismatch warning

So what I commonly see is that an interface-connection-macro gets created which relies on generate-statements and hierarchical references to the parameter values in order to hook up the testbench interface bits to the DUT and avoid any errors or warnings.

e.g.
```
`define connect_if(DUT, INTF) \
    for (genvar i = 0; i < 32; i++) begin: SIGNAL_CONNECT \
        if (i < DUT.MAX_WIDTH) begin: VALID \
            assign INTF.signal[i] = DUT.signal[i]; \
        end else begin: INVALID \
            assign INTF.signal[i] = 1'b0; \
        end \
    end
```

MikePopoloski, 2022-01-28:
Support for this has been added. Interface port connections are now allowed to be hierarchical. Hierarchical references to constants are allowed in constant expressions if you pass the `--allow-hierarchical-const` flag. I also added a `--compat vcs` flag as a shortcut for options that increase compatibility with VCS.

The new features are lightly tested; please open additional issues if you encounter problems with them.

closed 2022-01-28 by MikePopoloski