MikePopoloski/slang #433: part-select out of bound , is it a bug ?

aegean0228, 2021-09-14:
module tb();
    parameter reg[1:0] P = 2'b11;
    parameter reg[2:0] P1 = P[2:0];
    endmodule

i got: terminate called after throwing an instance of 'std::bad_variant_access'
  what():  Unexpected index
Aborted (core dumped)

LRM said: Part-selects that are partially out of
range shall, when read, return x for the bits that are out of range

MikePopoloski, 2021-09-14:
When I try your example I get:
```
source.sv:3:27: error: cannot select range of [2:0] from 'reg[1:0]'
parameter reg[2:0] P1 = P[2:0];
                          ^~~
```
Can you post the actual code that results in the exception?

aegean0228, 2021-09-14:
i use slang as a library 
auto tree = SyntaxTree::fromText(R"(
module tb();
parameter reg[1:0] P = 2'b11;
parameter reg[2:0] P1 = P[2:0];
endmodule
)");
then i did something like:
    Compilation compilation;
    auto& instance = evalModule(tree, compilation).body;
     auto& i1 = instance.find<ParameterSymbol>("P1");

MikePopoloski, 2021-09-14:
Ah, from a library context you can't assume the result type of a constant evaluation, you have to check what kind of value it holds. But I take your real point is whether the code should be allowed to compile or report an error; the LRM is clear about out-of-bounds accesses for runtime code, less clear about in a constant expression context. I chose to make slang report an error as this is clearly a coding mistake and it's nice to catch it at compile time, but I suppose you could make an argument that it should just be a warning and be allowed to compile as x's. It's not particularly hard to make that change.

aegean0228, 2021-09-14:
Yes, i believe VCS will report warning in this case. Anyway it's not an urgent issue, just let you know there is behavior difference for out of bound access

MikePopoloski, 2022-08-31:
I made these errors suppressible warnings, which will be the default when running in vcs compat mode. Otherwise they will default to still being an error, as the LRM allows:
> NOTE 2—Bit-select or part-select indices that are outside the declared range may be flagged as a compile time error.

closed 2022-08-31 by MikePopoloski