MikePopoloski/slang #434: Real math function unittest failure for SLES11.4 build

jrudess, 2021-09-16:
I'm working on deploying slang in a corporate environment using SLES11.4, and the build is successful but one unit test fails.
```
slang/tests/unittests/EvalTests.cpp:1522: FAILED:
  CHECK( session.eval("$tan(123.456)").real() == tan(123.456) )
with expansion:
  1.35181 == 1.3518050822
```
Tool versions are:
```
cmake 3.15.2
gcc 9.2
python 3.6
```
gcc seems like the most likely culprit, but wanted to check if you have any other suggestions.

MikePopoloski, 2021-09-16:
Yeah, I assume gcc and libc are compiled with differing rounding modes or something here. I can change these tests to check for equality within epsilon to make them more stable.

MikePopoloski, 2021-09-17:
Fixed in c5b971818bb8b54935c40a5bb54432764e5f1ce4

closed 2021-09-17 by MikePopoloski