MikePopoloski/slang #437: Using a bit-slice of a rand-variable does not work for dist constraints

jrudess, 2021-09-16:
This example throws an error for a dist-constraint using a bit-slice, but it does work for an inside-constraint.
```
class C;
    rand bit [63:0] value;
    constraint value_c {
        value[63] dist {0 :/ 70, 1 :/ 30};
        value[0] == 1'b0;
        value[15:8] inside {
            8'h0,
            8'hF
        };
    }
    function new();
    endfunction
endclass
```
```
slangtest2.sv:4:9: error: 'dist' expression must reference at least one 'rand' variable
        value[63] dist {0 :/ 70, 1 :/ 30};
        ^~~~~~~~~
```

MikePopoloski, 2021-09-16:
Yep, this is a bug. Thanks for the report.

MikePopoloski, 2021-09-17:
Fixed in 445da5938cea6c6ba77947b051aa48c5761dccab

closed 2021-09-17 by MikePopoloski