MikePopoloski/slang #440: For an array of events, the individual events cannot be triggered

jrudess, 2021-09-16:
```
class C;
    protected event my_event[8];
    function new();
    endfunction
    task process_txn(int i);
        // ... do some work here  then trigger event ...
        ->my_event[i];
    endtask
endclass
```
```
slangtest3.sv:7:11: error: expression is not a triggerable event
        ->my_event[i];
          ^~~~~~~~~~~
```

MikePopoloski, 2021-09-16:
I think this is another case that is not technically legal syntax according to the grammar, but seems like an oversight because this should surely be allowed.

jrudess, 2021-09-16:
I can't test with Questa/Xcelium to see if either prints any warnings, but I had the same concern about LRM compliance.  I considered a fairly simple workaround that passes cleanly.
```
class C;
    typedef struct {
        event my_event;
    } my_struct_t;
    protected my_struct_t events[8];
    function new();
    endfunction
    task process_txn(int i);
        // ... do some work here  then trigger event ...
        ->events[i].my_event;
    endtask
endclass
```

MikePopoloski, 2021-09-17:
Fixed in be3f40fa91ce24449f34e35a30ebd531fec6de49

closed 2021-09-17 by MikePopoloski