MikePopoloski/slang #442: enum typedef scoping between a parent and child class

jrudess, 2021-09-16:
VCS appears to scope enum typedefs defined within a class to only be scoped for that exact class.  All other scopes, including children classes, must use a scoping operator to reference the enum value, e.g. `A::DATA`.  

Slang appears to inherit the typedef from the parent class into the child class resulting in a redefinition error at the enum label. It's not clear to me yet what the LRM intention is here.

```
class A;
    typedef enum int {
        DATA = 0
    } field_type_t;
endclass
class B extends A;
    typedef enum int {
        DATA = 3
    } index_t;
endclass
```

At any rate, I do tend to consider this a poor coding practice, and plan to change the code to not reuse the label in the child class.

MikePopoloski, 2021-09-18:
Typedefs are supposed to be inherited, and indeed VCS and slang both support that. However, they're also supposed to be able to be overridden in the derived class; slang supports that for the typedef itself but I overlooked it for the enumerands.

MikePopoloski, 2021-09-18:
Fixed in cbebae399a6e3b4057235aed0270ffa9e80031ac

closed 2021-09-18 by MikePopoloski