MikePopoloski/slang #443: Handling of vendor-specific system tasks/functions

jrudess, 2021-09-16:
VCS supports a system function called $systemf() which can be used to execute a shell command.  This is outside of the LRM.

To avoid parser errors for projects utilizing this VCS-specific feature, could there be a command-line or config-file mechanism for defining a user system task/function which is more lax about argument and return types, just to avoid the parser errors?  I am also considering whether to wrap the logic inside an `ifdef VCS` but not set that +define for slang parsing.

MikePopoloski, 2021-09-17:
I'm not opposed to adding vendor-specific functions, maybe hiding them behind a compiler flag that enables a compatibility mode or something. Having some kind of "blessed" whitelist of system names to essentially just ignore also seems doable although a little weird; just doing an ifdef might be easier at that point.

jrudess, 2021-09-17:
It turns out that both OVM and UVM use a vendor-specific function $psprintf().   This function originally came from VCS and was proposed to the SV Committee for adoption in to IEEE 1800.  The committee accepted the function but rejected the name Synopsys chose, and so the LRM name is $sformatf.   (What I heard was that the p in psprintf stood for perl, which is why the name was rejected).

Unfortunately, $psprintf() existed for a long time before $sformatf() made it to the standard and so it seems to be ubiquitous.   This might be one case where support of a vendor-specific function is warranted.   I would assume that Questa and Xcelium both also support this function given that it's in OVM and UVM.

MikePopoloski, 2021-09-17:
It's easy enough to add $psprintf as an alias for $sformatf, assuming all the arguments are the same. I'll add a warning that's only enabled as part of -Wpedantic that it's a non-standard function.

MikePopoloski, 2021-09-18:
Added in b675fa5d2764a31e14965954449a39ed70dde253

closed 2021-09-18 by MikePopoloski