MikePopoloski/slang #447: Problems accessing clocking-blocks from virtual interfaces

jrudess, 2021-09-17:
Looking at chapter 14 of the LRM, it doesn't seem to show any examples of accessing clocking blocks from a virtual-interface, but only from modules.  The use of clocking-blocks in classes via virtual-interfaces is necessary in order for a UVM BFM to resolve delta-delay race issues.
```
interface intf(
    input logic        clk,
    input logic        rst,
    input logic [31:0] data
);
    clocking cb @(posedge clk);
        default input #1step;
        input rst;
        input data;
    endclocking
endinterface
class A;
    virtual intf _my_intf;
    logic [31:0] dataQ[$];
    function new();
    endfunction
    task sample_data();
        @(_my_intf.cb);
        dataQ.push_back(_my_intf.cb.data);
    endtask
endclass
```
```
slangtest8.sv:18:19: error: cannot access 'cb' in 'intf' with '.'
        @(_my_intf.cb);
          ~~~~~~~~^~~
slangtest8.sv:19:33: error: cannot access 'cb' in 'intf' with '.'
        dataQ.push_back(_my_intf.cb.data);
                        ~~~~~~~~^~~~~~~~
```

MikePopoloski, 2021-09-17:
Yeah, the section you should look at is 25.9.1 which discusses clocking blocks and virtual interfaces. Now that you've mentioned it I'm reminded that this is one of those things I haven't looked at yet and is contributing to virtual interface support being "partial".

MikePopoloski, 2021-09-28:
Should be fixed in 18a2736010e251bd77743a2acb8eb142c9cc8c5e

closed 2021-09-28 by MikePopoloski

jrudess, 2021-09-29:
Thanks!  This was the last feature needed to get my first repo (verification BFM, no RTL) parsing cleanly with slang.

MikePopoloski, 2021-09-29:
Hurray!