MikePopoloski/slang #451: sigsegv for Diagnostics

jrudess, 2021-09-17:
I'm seeing a diagnostic segfault on on the 325th error for a larger project (determined by playing around with --error-limit).  The error doesn't print, so I can't easily create a minimal test-case.

Debug build assertion:
```
internal compiler error: Assertion 'location' failed
  in file slang/source/diagnostics/Diagnostics.cpp, line 77
  function: slang::Diagnostic& slang::Diagnostics::add(slang::DiagCode, slang::SourceLocation)
```

I've been unable to collect a useful backtrace with gdb.   I think this may be the result of having such ancient software (SLES11.4).
```
Dwarf Error: wrong version in compilation unit header (is 4, should be 2) [in module slang/build/bin/slang]
(gdb) run
Starting program: slang/build/bin/slang --error-limit=325  (... lots of -D, -I, files here ... )
Top level design unit:
     top

[New Thread 0x7ffff7fbb7820 (LWP 104817)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7fbb720 (LWP 1040817)]
(gdb) backtrace
#0  0x00000000005f3940
```
Do you have any recommendations on how to help narrow it down?  I'm working on getting access to some SLES12 machines which have (slightly) newer software.

MikePopoloski, 2021-09-17:
Any backtrace from gdb would tell you right away, so I would try to get that working. The assertion is because I require all issued diagnostics to have a valid location and for whatever reason one was issued with an empty location. That's a bug that should be fixed but more context needed to know where that's happening.

jrudess, 2021-09-17:
Rebuilding with -gdwarf-2 resolved the backtrace collection.   The redacted1.sv file only contains SV properties used by assertions elsewhere in the project.   

To reproduce the segfault with the debug build, I commented out two assertions.
```
Diagnostics.cpp:77
MiscExpressions.cpp:814
```
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7fbb720 (LWP 169342)]
0x00000000005f3940 in slang::AssertionInstanceExpression::bindPort (symbol=@0xc21f488, range=
        {startLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, charOf
fset = 16767}, endLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, ch
arOffset = 16777}}, instanceCtx=@0x7fffffff1f10) at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
318               return 0;

(gdb) backtrace
#0  0x00000000005f3940 in slang::AssertionInstanceExpression::bindPort (symbol=@0xc21f488, range=
        {startLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, charOf
fset = 16767}, endLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, ch
arOffset = 16777}}, instanceCtx=@0x7fffffff1f10) at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
#1  0x00000000005cce34 in slang::Expression::bindLookupResult (compilation=@0x7fffffff5070, result=@0x7fffffff1b90, syntax=@0xaafc890, invocation=0x0, withClause=0x0, context=@0x7fffffff1f10)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#2  0x00000000005cc84d in slang::Expression::bindName (compilation=@0x7fffffff5070, syntax=@0xaafc890, invocation=0x0, withClause=0x0, context=@0x7fffffff1f10)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#3  0x00000000005cc105 in slang::Expression::create (compilation=@0x7fffffff5070, syntax=@0xaafc890, ctx=@0x7fffffff2190, extraFlags={static mask_value = <optimized out>, m_bits = 16}, assignmentTarget=0x0)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#4  0x00000000005d1966 in slang::Expression::selfDetermined (compilation=@0x7fffffff5070, syntax=@0xaafc890, context=@0x7fffffff2190, extraFlags={static mask_value = <optimized out>, m_bits = 16})
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#5  0x00000000005c7733 in slang::Expression::bind (syntax=@0xaafc890, context=@0x7fffffff2190, extraFlags={static mask_value = <optimized out>, m_bits = 16})
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#6  0x000000000068bb82 in slang::Builtins::BitsFunction::bindArgument (this=0xb74380, context=@0x7fffffff2240, syntax=@0xaafc890)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#7  0x00000000005a4d49 in slang::CallExpression::createSystemCall (compilation=@0x7fffffff5070, subroutine=@0xb74380, firstArg=0x0, syntax=0xaafc978, withClause=0x0, range=
        {startLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, charOf
fset = 16761}, endLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, ch
arOffset = 16778}}, context=@0x7fffffff2f60, randomizeScope=0x0) at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
#8  0x00000000005a228d in slang::CallExpression::fromLookup (compilation=@0x7fffffff5070, subroutine=@0x7fffffff2df0, thisClass=0x0, syntax=0xaafc978, withClause=0x0, range=
        {startLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, charOf
fset = 16761}, endLoc = {bufferName = {static npos = <optimized out>, _M_len = 48, _M_str = 0x9016990 "<redacted1>.sv"}, static NoLocation = <optimized out>, bufferID = 52284, ch
arOffset = 16778}}, context=@0x7fffffff2f60) at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
#9  0x00000000005cc80c in slang::Expression::bindName (compilation=@0x7fffffff5070, syntax=@0xaafc810, invocation=0xaafc978, withClause=0x0, context=@0x7fffffff2f60)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#10 0x00000000005a21a7 in slang::CallExpression::fromSyntaxImpl (compilation=@0x7fffffff5070, left=@0xaafc810, invocation=0xaafc978, withClause=0x0, context=@0x7fffffff2f60)
    at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
#11 0x00000000005a1f7a in slang::CallExpression::fromSyntax (compilation=@0x7fffffff5070, syntax=@0xaafc978, withClause=0x0, context=@0x7fffffff2f60)
    at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
#12 0x00000000005cbbc2 in slang::Expression::create (compilation=@0x7fffffff5070, syntax=@0xaafc978, ctx=@0x7fffffff3200, extraFlags={static mask_value = <optimized out>, m_bits = 0}, assignmentTarget=0x0)
    at gcc/9.2.0/include/c++/9.2.0/bits/stl_deque.h:95
#13 0x0000000000600892 in slang::BinaryExpression::fromSyntax (compilation=@0x7fffffff5070, syntax=@0xaafca58, context=@0x7fffffff3200)
    at gcc/9.2.0/include/c++/9.2.0/bits/char_traits.h:318
[... cut ...]
#46 0x0000000000457df3 in slang::Compilation::getSemanticDiagnostics (this=0x7fffffff5070) at gcc/9.2.0/include/c++/9.2.0/typeinfo:100
#47 0x00000000004589bd in slang::Compilation::getAllDiagnostics (this=0x7fffffff5070) at gcc/9.2.0/include/c++/9.2.0/typeinfo:100
#48 0x000000000040c5b4 in Compiler::run (this=0x7fffffff4db0) at gcc/9.2.0/include/c++/9.2.0/typeinfo:100
#49 0x00000000004122bc in driverMain<char**> (argc=69, argv=0x7fffffff6778, suppressColorsStdout=false, suppressColorsStderr=false) at gcc/9.2.0/include/c++/9.2.0/typeinfo:100
#50 0x0000000000409c4a in main (argc=69, argv=0x7fffffff6778)
```

MikePopoloski, 2021-09-18:
I think this is two separate issues here. The first assertion inside Diagnostic is telling us that a proper diagnostic was being issued but the code didn't provide a valid location for it. A backtrace on that assert should tell us which code has that bug.

The second assert in bindPort() is saying some internal logic is wrong whereby we managed to look up an assertion port without actually being in the middle of expanding an assertion, which shouldn't be possible. Commenting out that assert just ensures you'll crash because the context pointer is null. I think I need some example source code that led to this being hit.

jrudess, 2021-09-18:
Sure thing.  Will try to get a backtrace at the point of the first assertion and also see if I can minimize the code down.

jrudess, 2021-09-19:
Here's a stacktrace at the ASSERT just before it fires.  May still not be too useful, but might help me narrow down where to find the SV-assertion that I believe is the root cause since `parseIfGeneratedConstruct` is in the backtrace.
```c++
#0  slang::Diagnostics::add (this=0x7fffffff2950, code=..., location=...) at slang/source/diagnostics/Diagnostics.cpp:80
#1  0x000000000080b541 in slang::Lexer::addDiag (this=0x7fffffff26e0, code=..., offset=1) at slang/source/parsing/Lexer.cpp:1107
#2  0x0000000000809f59 in slang::Lexer::lexStringLiteral (this=0x7fffffff26e0) at slang/source/parsing/Lexer.cpp:630
#3  0x0000000000808be8 in slang::Lexer::lexToken (this=0x7fffffff26e0, keywordVersion=slang::KeywordVersion::v1800_2017) at slang/source/parsing/Lexer.cpp:232
#4  0x00000000008088f6 in slang::Lexer::lex (this=0x7fffffff26e0, keywordVersion=slang::KeywordVersion::v1800_2017) at slang/source/parsing/Lexer.cpp:185
#5  0x0000000000808020 in slang::Lexer::stringify (alloc=..., location=..., trivia=..., begin=0x7fffffff2e28, end=0x7fffffff2e28) at slang/source/parsing/Lexer.cpp:122
#6  0x000000000073c2a7 in slang::Preprocessor::applyMacroOps (this=0x7fffffff5a10, tokens=..., dest=...) at slang/source/parsing/Preprocessor_macros.cpp:235
#7  0x000000000073b5b6 in slang::Preprocessor::handleTopLevelMacro (this=0x7fffffff5a10, directive=...) at slang/source/parsing/Preprocessor_macros.cpp:80
#8  0x000000000072d7e2 in slang::Preprocessor::handleMacroUsage (this=0x7fffffff5a10, directive=...) at slang/source/parsing/Preprocessor.cpp:542
#9  0x000000000072b63d in slang::Preprocessor::handleDirectives (this=0x7fffffff5a10, token=...) at slang/source/parsing/Preprocessor.cpp:210
#10 0x000000000072b22a in slang::Preprocessor::nextProcessed (this=0x7fffffff5a10) at slang/source/parsing/Preprocessor.cpp:176
#11 0x00000000007312d6 in slang::Preprocessor::peek (this=0x7fffffff5a10) at slang/source/parsing/Preprocessor.cpp:1195
#12 0x0000000000731316 in slang::Preprocessor::consume (this=0x7fffffff5a10) at slang/source/parsing/Preprocessor.cpp:1200
#13 0x000000000072b1a6 in slang::Preprocessor::next (this=0x7fffffff5a10) at slang/source/parsing/Preprocessor.cpp:150
#14 0x0000000000728b76 in slang::ParserBase::Window::addNew (this=0x7fffffff5628) at slang/source/parsing/ParserBase.cpp:232
#15 0x0000000000727fe4 in slang::ParserBase::peek (this=0x7fffffff5620) at slang/source/parsing/ParserBase.cpp:55
#16 0x000000000070c873 in slang::Parser::parseMemberList<slang::MemberSyntax, slang::Parser::parseGenerateBlock()::<lambda(slang::SyntaxKind, bool&)> >(slang::TokenKind, slang::Token &, slang::SyntaxKind, slang::Parser::<lambda(slang::SyntaxKind, bool&)> &) (this=0x7fffffff5620, endKind=slang::TokenKind::EndKeyword, endToken=..., parentKind=slang::SyntaxKind::GenerateBlock, parseFunc=...)
    at slang/source/parsing/Parser_members.cpp:377
#17 0x00000000006fe381 in slang::Parser::parseGenerateBlock (this=0x7fffffff5620) at slang/source/parsing/Parser_members.cpp:844
#18 0x00000000006fda10 in slang::Parser::parseIfGenerateConstruct (this=0x7fffffff5620, attributes=...) at slang/source/parsing/Parser_members.cpp:757
#19 0x00000000006fa88a in slang::Parser::parseMember (this=0x7fffffff5620, parentKind=slang::SyntaxKind::GenerateRegion, anyLocalModules=@0x7fffffff45bf: false)
    at slang/source/parsing/Parser_members.cpp:235
#20 0x00000000006f9d1d in slang::Parser::<lambda(slang::SyntaxKind, bool&)>::operator()(slang::SyntaxKind, bool &) const (__closure=0x7fffffff47d8, parentKind=slang::SyntaxKind::GenerateRegion, 
    anyLocalModules=@0x7fffffff45bf: false) at slang/source/parsing/Parser_members.cpp:112
#21 0x000000000070b9a5 in slang::Parser::parseMemberList<slang::MemberSyntax, slang::Parser::parseMember(slang::SyntaxKind, bool&)::<lambda(slang::SyntaxKind, bool&)> >(slang::TokenKind, slang::Token &, slang::SyntaxKind, slang::Parser::<lambda(slang::SyntaxKind, bool&)> &) (this=0x7fffffff5620, endKind=slang::TokenKind::EndGenerateKeyword, endToken=..., parentKind=slang::SyntaxKind::GenerateRegion, parseFunc=...)
    at slang/source/parsing/Parser_members.cpp:381
#22 0x00000000006f9f11 in slang::Parser::parseMember (this=0x7fffffff5620, parentKind=slang::SyntaxKind::ModuleDeclaration, anyLocalModules=@0x7fffffff4c5f: false)
    at slang/source/parsing/Parser_members.cpp:113
#23 0x00000000006f995d in slang::Parser::<lambda(slang::SyntaxKind, bool&)>::operator()(slang::SyntaxKind, bool &) const (__closure=0x7fffffff4db8, parentKind=slang::SyntaxKind::ModuleDeclaration, 
    anyLocalModules=@0x7fffffff4c5f: false) at slang/source/parsing/Parser_members.cpp:64
#24 0x000000000070b799 in slang::Parser::parseMemberList<slang::MemberSyntax, slang::Parser::parseModule(slang::Parser::AttrList, slang::SyntaxKind, bool&)::<lambda(slang::SyntaxKind, bool&)> >(slang::TokenKind, slang::Token &, slang::SyntaxKind, slang::Parser::<lambda(slang::SyntaxKind, bool&)> &) (this=0x7fffffff5620, endKind=slang::TokenKind::EndModuleKeyword, endToken=..., 
    parentKind=slang::SyntaxKind::ModuleDeclaration, parseFunc=...) at slang/source/parsing/Parser_members.cpp:381
#25 0x00000000006f9b31 in slang::Parser::parseModule (this=0x7fffffff5620, attributes=..., parentKind=slang::SyntaxKind::CompilationUnit, anyLocalModules=@0x7fffffff543f: false)
    at slang/source/parsing/Parser_members.cpp:65
#26 0x00000000006fa0a6 in slang::Parser::parseMember (this=0x7fffffff5620, parentKind=slang::SyntaxKind::CompilationUnit, anyLocalModules=@0x7fffffff543f: false)
[... cut ...]
#32 0x0000000000408edc in loadAllSources (compilation=..., sourceManager=..., buffers=std::vector of length 5, capacity 8 = {...}, options=..., singleUnit=false, onlyLint=false, 
    libraryFiles=std::vector of length 0, capacity 0, libDirs=std::vector of length 0, capacity 0, libExts=std::vector of length 0, capacity 0)
    at slang/tools/driver/driver.cpp:129
#33 0x0000000000411fd2 in driverMain<char**> (argc=16, argv=0x7fffffff80a8, suppressColorsStdout=false, suppressColorsStderr=false) at slang/tools/driver/driver.cpp:690
#34 0x0000000000409c4a in main (argc=16, argv=0x7fffffff80a8) at slang/tools/driver/driver.cpp:809
```

jrudess, 2021-09-19:
Minimized sigsegv example
```systemverilog
`define default_clk 1'b0
property prop(a, b, c, d, clk=`default_clk, rst=1'b0);
    logic [$bits(b)-1:0] e;
    @(clk) disable iff(rst)
    (a, e = b) ##0
      (c or (!c ##1 (!a through d[->1])))
            |-> e == c);
endproperty: prop
```

MikePopoloski, 2021-09-20:
Thanks, the segsegv is easily reproducible with that snippet.

I have a general idea of what's causing that diagnostic assert -- something to do with stringifying via the preprocessor, but having a reproducer would be very helpful. If you have trouble tracking it down you can look at the backtrace when you hit that assert and try to find a SourceLocation object anywhere on the stack; in DEBUG builds that should contain the file name and byte offset of general area where the problem is happening.

jrudess, 2021-09-21:
It turns out that the code-example from #455 is the reproducer for the first ASSERT.  I hadn't noticed this because previously I was not running my test cases using a Debug build.   I'll start primarily running with debug build going forward to help weed issues out.

MikePopoloski, 2021-09-21:
The crash should be fixed as of a3c6a29f9fce359972838764cf6551c63bf473a7

MikePopoloski, 2021-09-22:
I can't reproduce the assert with the code examples in #455 -- are you sure that's the reproducer?

jrudess, 2021-09-22:
It looks like it might be gcc version dependent.   On SLES11 using gcc 9.2.0 the example reproduces.   On my home system using gcc 11.1.0 it doesn't.

MikePopoloski, 2021-09-24:
Hmm, so I tried with gcc 8.4 locally (on Ubuntu 20.04) and haven't reproduced the issue. Just to confirm, the code that causes it is as follows?
```
`define MAKE_INST(NAME, SIG) \
    (* instance_name = `"inst_``NAME`" *) \
    mod \inst_``NAME (.sig(SIG));

module mod(input logic sig);
endmodule
module top;
    logic sig1, sig2;
    `MAKE_INST(A, sig1)
    `MAKE_INST(B, sig2)
endmodule
```

jrudess, 2021-09-24:
Ack, sorry, I have found my mistake. The example in #455 passes because it has a slight syntax difference with continuations I was glossing over. Here is one that should fail and it's probably nothing to do with gcc versions. The only change is for the module name and signal list to be split out onto their own lines which is more common syntax when there are lots of signals.

```systemverilog
`define MAKE_INST(NAME, SIG) \
    (* instance_name = `"\inst_``NAME`" *) \
    mod \inst_``NAME ( \
        .sig(SIG) \
    );

module mod(input logic sig);
endmodule
module top;
    logic sig1, sig2;
    `MAKE_INST(A, sig1)
    `MAKE_INST(B, sig2)
endmodule
```

MikePopoloski, 2021-09-24:
Ok yeah, that reproduces it.

MikePopoloski, 2021-09-25:
Assert should be fixed in cac1a4df9afc5dc46491ab03a179bfe88605b950

closed 2021-09-25 by MikePopoloski