MikePopoloski/slang #455: Macro does not expand arguments inside strings using `"

jrudess, 2021-09-18:
This isn't something I've run into before, but it looks like `" is used for allowing macro arguments to expand while inside quotation marks ($22.5.1).  Based on the error, I think the macro expansion is not occurring in this case.

```
`define MAKE_INST(NAME, SIG) \
    `ifdef MACRO_ATTRIBUTE \
        (* macro_attribute = `"MAKE_INST(NAME, SIG))`" *) \
    `endif \
    (* instance_name = `"inst_``INST`" *) \
    mod inst_``INST(.sig(SIG));

module mod(input logic sig);
endmodule
module top;
    logic sig1, sig2;
    `MAKE_INST(A, sig1)
    `MAKE_INST(B, sig2)
endmodule
```
```
slangtest17.sv:13:5: error: redefinition of 'inst_INST'
    `MAKE_INST(B, sig2)
    ^
slangtest17.sv:6:9: note: expanded from macro 'MAKE_INST'
    mod inst_``INST(.sig(SIG));
        ^
slangtest17.sv:12:5: note: previous definition here
    `MAKE_INST(A, sig1)
    ^
slangtest17.sv:6:9: note: expanded from macro 'MAKE_INST'
    mod inst_``INST(.sig(SIG));
```

jrudess, 2021-09-18:
Sorry, this analysis is wrong and the errors are from a typo.  Will reopen when I"m able to reproduce the original failure in a minimized example.

jrudess, 2021-09-18:
Original example updated to be correct.

MikePopoloski, 2021-09-18:
Hmm, this makes no sense to me. The backslash starts an escaped identifier, which only terminates at the next whitespace character, so the name here is "inst_``NAME". This doesn't match any of the macro argument names, of course. I guess this is saying that within every escaped identifier you need to look for any substrings that contain macro backticks and try splitting them and doing argument substitution?

jrudess, 2021-09-18:
I've tracked down why the RTL requires the escape character.  It is instantiating a module multiple times within a generate-for-loop using the macro.   It is naming the module with brackets and the genvar-loop variable to give each module a unique name.  But because the module name contains brackets, the escape character is required for the entire string.

```
localparam MAX=8;
logic signal[MAX-1:0];
logic sig1;
for (genvar index = 0; index < MAX; index++) begin: GEN_MOD
    `MAKE_INST(signal[index], sig1)
```

In the hierarchy, there should then be the following instances.
```
top.GEN_MOD[0].inst_signal[0]
top.GEN_MOD[1].inst_signal[1]
...
top.GEN_MOD[7].inst_signal[7]
```

This seems totally unnecessary to me, since the generate-label is what provides the unique module hierarchy, but VCS does accept this.

jrudess, 2021-09-18:
GIven that macro-expansion does not work on quotations by default, necessitating the `" special syntax, I think there's a strong argument to be made that VCS is not LRM compliant here.

MikePopoloski, 2021-09-18:
I checked The Big Three and they all support this, so... 🤷 I guess I will too.

MikePopoloski, 2021-09-25:
Added support in 2065f413a8125992d0356ad60140aa5f0c510a2d

closed 2021-09-25 by MikePopoloski