MikePopoloski/slang #456: Assert at AssignmentExpressions.cpp, line 705 for malformed types in dist constraint

jrudess, 2021-09-21:
The code shown below is malformed and the rand-variable is using an incorrect data-type (type should have been `int unsigned`).  The code parses without error or warning for a Release build, but a Debug build fires the following assertion.

```c++
internal compiler error: Assertion 'targetType.isAssignmentCompatible(*expr.type) || ((targetType.isString() || targetType.isByteArray()) && expr.isImplicitString()) || (targetType.isEnum() && isSameEnum(expr, targetType))' failed
  in file slang/source/binding/AssignmentExpressions.cpp, line 705
  function: static slang::Expression& slang::ConversionExpression::makeImplicit(const slang::BindContext&, const slang::Type&, slang::ConversionKind, slang::Expression&, slang::SourceLocation)
```

```systemverilog
class C;
    typedef enum {
        ENUM_A,
        ENUM_B,
        ENUM_C
    } my_enum_t;
    rand my_enum_t arr[2][3][8];
    constraint arr_c {
        foreach (arr[i, j, k]) {
            arr[i][j][k] dist {
                [0:3]   :/ 65,
                [4:7]   :/ 20,
                [8:11]  :/ 19,
                [12:15] :/ 5
            };
        }
    }
endclass

```

jrudess, 2021-09-21:
I should also note that VCS gives no error for this example, and slang is really proving its value having found quite a few bugs in production code that VCS misses.

MikePopoloski, 2021-09-21:
Thanks, that's great to hear. I'll look into the assert, but the code isn't actually invalid right? Like you might want a warning about it but it's valid to pick out bits from an enum variable.

jrudess, 2021-09-21:
Yeah, just reread the LRM and it's not invalid.  I had thought that values beyond the range of the defined labels (ENUM_A=0, ENUM_B=1, ENUM_C=2) was considered an error, but values only have to be within the range of the type (int for this case).  But a warning for values that don't match any declared ENUM value would be a great addition to catch coding errors.

jrudess, 2021-09-24:
Ack, sorry, I have found my mistake.   The example in #455 passes because it has a slight syntax difference with continuations I was glossing over.   Here is one that should fail and it's probably nothing to do with gcc versions.  The only change is for the module name and signal list to be split out onto their own lines which is more common syntax when there are lots of signals.

```systemverilog
`define MAKE_INST(NAME, SIG) \
    (* instance_name = `"\inst_``NAME`" *) \
    mod \inst_``NAME ( \
        .sig(SIG) \
    );

module mod(input logic sig);
endmodule
module top;
    logic sig1, sig2;
    `MAKE_INST(A, sig1)
    `MAKE_INST(B, sig2)
```

MikePopoloski, 2021-09-25:
Fixed in c36e643572d95f0a0c216e7637540fbe3642ec23

closed 2021-09-25 by MikePopoloski