MikePopoloski/slang #458: Compiler indicator statements are not supported

YJT13845, 2021-09-23:
The synopsys translate_off and synopsys translate_on are not supported.
example:
[image]

MikePopoloski, 2021-09-23:
Yes, slang is not built by synopsys. Also putting directives to the compiler inside comments is pretty much the worst idea I've ever seen implemented in a programming language and I have no desire to perpetuate it. This problem is easily solved with an ifdef or a conditional generate instead, and it will make your code portable to all tools instead of being tied to a single one.

YJT13845, 2021-09-24:
You have a point there. Thank you for your reply!

closed 2021-09-24 by YJT13845