MikePopoloski/slang #459: parameter overrides don't work

aegean0228, 2021-09-23:
module tb();
    dut #(.P1(1), .P2(1), .P3(1)) inst();
endmodule
module dut();
    reg data;
    function int getv();
        return 1;
    endfunction
    parameter P0 = $bits(data);
    parameter [P0:0] P1 = 0;
    parameter [getv():0] P2 = 0;
    parameter reg [$bits(data)] P3 = 0;
endmodule

give below errors:

<unnamed_buffer0>:69:16: error: use of undeclared identifier 'P0'
    parameter [P0:0] P1 = 0;
               ^~
<unnamed_buffer0>:70:16: error: use of undeclared identifier 'getv'
    parameter [getv():0] P2 = 0;
               ^~~~
<unnamed_buffer0>:71:26: error: use of undeclared identifier 'data'
    parameter reg [$bits(data)] P3 = 0;
                         ^~~~
//code
createParams
     // Need the imports here as well, since parameters may depend on them.
    for (auto import : definition.syntax.header->imports)
        tempDef.addMembers(*import);

    return paramBuilder.createParams(tempDef, ll, instanceLoc, forceInvalidParams,
                                     /* suppressErrors */ false);

    the type of the  value parameter could depend on nodes/functions/types/class/etc

amal-khailtash, 2021-09-29:
I am seeing this same issue, but unfortunately I cannot provide a testcase.  For some reason, it does not happen on a smaller design.  But when I have multiple modules/packages, it shows up.  The problem seems to be a module parameter getting its value from a package localparam where that localparam is included from another file.

MikePopoloski, 2021-09-29:
I am able to reproduce with the snippet above. This is a bug in how body parameters that depend on local members get their values.

MikePopoloski, 2022-01-01:
Should be fixed now.

closed 2022-01-01 by MikePopoloski