MikePopoloski/slang #462: std::randomize does not accept variables contained within a class or struct

jrudess, 2021-09-25:
VCS accepts these, but the LRM does state that std::randomize() operates on variables in the local scope.

```systemverilog
class C;
    bit [5:0] code;
endclass
typedef struct {
    bit [5:0] code;
} S;
class D;
    function void f();
        S s;
        C c = new();
        std::randomize(c.code) with {
        };
        std::randomize(s.code) with {
        };
    endfunction
endclass
```
```systemverilog
slangtest23.sv:11:24: error: expected variable name
        std::randomize(c.code) with {
                       ^~~~~~
slangtest23.sv:13:24: error: expected variable name
        std::randomize(s.code) with {
                       ^~~~~~
```

MikePopoloski, 2021-09-25:
This looks like another extension. The grammar specifies that std::randomize takes a variable_identifier_list which is just a comma separated list of identifiers, no room for fanciness.

jrudess, 2021-09-25:
Thanks, that's also the impression I got from the grammar.

closed 2021-09-25 by jrudess