MikePopoloski/slang #463: foreach parsing issues

jrudess, 2021-09-25:
This example parses cleanly prior to d155fd6
```systemverilog
class A;
    int subhash[string];
endclass
class C;
    A hash[string];
    task t(string index);
        foreach (hash[index].subhash[i]) begin
        end
    endtask
endclass
```
```systemverilog
slangtest24.sv:7:29: error: expected ')'
        foreach (hash[index].subhash[i]) begin
                            ^
slangtest24.sv:7:17: note: to match this '('
        foreach (hash[index].subhash[i]) begin
                ^
slangtest24.sv:7:29: error: expected statement
        foreach (hash[index].subhash[i]) begin
                            ^
slangtest24.sv:7:30: error: use of undeclared identifier 'subhash'
        foreach (hash[index].subhash[i]) begin
                             ^~~~~~~
slangtest24.sv:7:40: error: expected ';'
        foreach (hash[index].subhash[i]) begin
                                       ^
```

MikePopoloski, 2021-09-25:
Hmm, more extensions to the foreach syntax eh? Hard to anticipate all the ways it might be used when those ways are not allowed in the language grammar. The fact that it was allowed prior to my commit is an oversight, but I can make it work again -- it seems like you just want any arbitrary expression that evaluates to an array prior to the final brackets that declare loop variables?

jrudess, 2021-09-25:
I believe that's the behavior the big-three allow (e.g. arbitrary expressions that evaluate to an array).  I'm not sure if Questa reports a warning for this specific example like it does for some of the other foreach cases.  

But I have no issues if you want to reject changes here to stay aligned to the current LRM grammar.  I don't see any examples of this foreach-style coming from OVM or UVM in the latest parses, and I've already begun to move away from foreach loops in favor of traditional for-loops given the current LRM grammar.

MikePopoloski, 2021-09-25:
Fixed in e2a2513da847787c867fc688530989ffc45352b3

closed 2021-09-25 by MikePopoloski